dangiu/PicoMemcard #1: VBUS connected to 3.3v

darrena092, 2022-06-05:
Hey,

Just setting this up, and I've noticed you say to connect VBUS on the pico to 3.3v on the PS2, which creates the danger that when plugging in the pico to usb, you'd be connecting that to 5v.

You can just connect the 3.3v memory card pin to 3.3v on the pico, and it'll power it just fine. That'll also allow you to safely have the board plugged into the PC for debugging at the same time.

darrena092, 2022-06-05:
(Also just to note I'm testing this with a Playstation 2 - I guess in theory at least it should recognise the PS1 memory card)

dangiu, 2022-06-05:
The problem with this approach is that the Pico can be powered on VSYS using only 3.3V (it expects a voltage already regulated). Some playstation models are documented to output more than 3.3 volts on that line (up to 5 or 7 if i remember correctly) so connecting it to VSYS would risk damaging the Pico.

darrena092, 2022-06-05:
Ah fair enough, mine didn't blow up anyways 🤣 - I'll measure the 3.3v line on the ps1 tomorrow.

dangiu, 2022-06-05:
Okk go for it! Same goes for my PSOne Model SCPH-102, measured some time ago but I guess you can never be too sure 😄

darrena092, 2022-06-08:
3.45v, a little on the high side. Will probably just leave the 3v3 line disconnected and power it from the PC while I'm working on it. I don't think an extra 150mv is going to cause all that big of a problem, but might be a good idea if there's a v2 PCB to include some voltage regulation, even just an SMT LM317.

dangiu, 2022-06-09:
Fortunately the Pico already regulates the power provided on VBUS (since USB normally uses 5 volts) down to 3.3V so there's no need for additional components 😀

closed 2022-06-09 by dangiu