dangiu/PicoMemcard #2: Not detected by Playstation 2 - SCPH-39003

darrena092, 2022-06-05:
Hi,

Tested this today with my original fat playstation 1, worked perfectly (using the breakout board I talked about on the Reddit thread that I'd made a while back).

When trying it with my Playstation 2 (SCPH-39003), it wasn't detected at all. Not sure what the issue there is, but I'll try to debug it if I get time. I understand the primary purpose of this project is for use on the Playstation 1, but it'd be nice if it also worked in backwards compatibility mode on the PS2 also.

dangiu, 2022-06-05:
Ok! Glad to hear it works on FAT PS1, could you provide the exact model? I'm just curious to know. I'm not sure on the protocol the PS2 uses to communicate with memory card, maybe the interaction is slightly different and messes up the PIO state machines. 
If you have an oscilloscope it would simplify the debugging process a ton! I used another Raspberry Pi Pico together with [Scoppy](https://github.com/fhdm-dev/scoppy). Thanks for your contribution!

darrena092, 2022-06-05:
The PS1 is a SCPH-9002 btw

darrena092, 2022-06-08:
This might be simpler than what you think.

Getting the following over UART:
```
SYNC    OK

Unknown MC CMD: 11

Unknown MC CMD: f3
WRITE  003f
```

Looks like there are just a few unimplemented commands that the PS2 sends, which the PS1 does not. I'll work on figuring out what they are and get them handled.

darrena092, 2022-06-08:
Nah scratch that, something weird going on. Going to hook up the logic analyser.

darrena092, 2022-06-08:
I have a theory, those 0x11 and 0xF3 commands aren't actually real. They seem to be generated when the PS2 does this:

[image]

I unfortunately don't have a real ps1 memory card to test with at the minute, but I'd imagine this must be part of some sort of mechanism to figure out whether or not the memory card is a PS2 or PS1 one.

The issue might be ACKing that. Hard to say without having a real one to test with at the minute, but I'd say we probably only want to ACK on /SEL with 8 clock pulses. I'll see if I can modify it to do that and report back.

uyjulian, 2022-07-08:
For reference, if the card inserted was no card or PS2 card before, it will attempt to probe in this order: PS2, PS1, PocketStation

Otherwise, it will probe in this order: PS1, PS2, PocketStation

Reference: https://github.com/ps2dev/ps2sdk/blob/f5401e0c95e85900464c550060da4c9b9ee6b67b/iop/memorycard/mcman/src/main.c#L497

darrena092, 2022-07-10:
Fixed in #10 (not yet merged) - though the fix is specific to the SD card version of the project due to problems with littlefs hard faulting the CPU with the new sector at a time sync process.

dangiu, 2022-08-02:
Fixed in new release

closed 2022-08-02 by dangiu