dangiu/PicoMemcard #3: With RP2040 zero，ps1 not detect

sceic, 2022-06-05:
With RP2040 zero, the .mcr file can be recognized through usb after writing the program. However, it cannot be detected by ps1 9002. It has been checked  3.3v power supply and the data line connection are not abnormal.

sceic, 2022-06-05:
[image]

[image]

darrena092, 2022-06-05:
How are you testing whether or not it detects it? Did you program it with an image containing saves and they don't show up in the bios?

Also, have you checked continuity between the edge connector on the memory card and the copper pads on the rp2040 zero board?

sceic, 2022-06-06:
How are you testing whether or not it detects it? Did you program it with an image containing saves and they don't show up in the bios?

A: Tested through the ps1 bios - memory card management interface, the slot 2 card cannot be copied to the Pico memorycard（slot 1）.

Also, have you checked continuity between the edge connector on the memory card and the copper pads on the rp2040 zero board?

Edit: Did you also cut the traces fully? It's hard to see from the photo. That could cause it to have a bad time too.

darrena092, 2022-06-06:
That's really interesting, so it works in a game, but not in the bios. You can't see your save icons in the memory card browser? In your first comment you said it's a SCPH-9002?

dangiu, 2022-06-06:
Very interesting indeed. SCPH-9002 is a PAL model running bios version 4.1 I believe. It would be interesting to test if other models with the same firmware version behave equally.
I have a couple of questions:
1. Do you have an oscilloscope? Could you capture the data being exchanged between PicoMemcard and PSX
2. In what game can you use it correcly to save/load? Could you try a couple different games?

sceic, 2022-06-07:
> Very interesting indeed. SCPH-9002 is a PAL model running bios version 4.1 I believe. It would be interesting to test if other models with the same firmware version behave equally. I have a couple [...]

I have an oscilloscope Rigol DS1102Z-E (100MHz bandwidth and 1GSa/s) but need guidance on how to test data exchange as I just bought it.

The game is FinalFantasy 8, other games have not been tested, I will try other games later, such as Tomb Raider 2.

sceic, 2022-06-07:
> That's really interesting, so it works in a game, but not in the bios. You can't see your save icons in the memory card browser? In your first comment you said it's a SCPH-9002?

In bio.  memory card browser no icons.  and can't copy saves betwin slot1&2. 

Yes,my PS1 is SCPH-9002.

darrena092, 2022-06-07:
Odd, I'm using the same console, SCPH-9002 and saves show up fine in the bios.

dangiu, 2022-06-07:
> 

I'm not familiar with that model of oscilloscope but I can give you some general advices:
- double all of the connection on the memory card side and hook them up to the oscilloscope
- look for something like "record fixed amount of samples"
- plug in your PicoMemcard and trigger the recording process measuring the communication
- desolder the RP2040 zero leads but leave the connection to the oscilloscope
- now plug in a working memory card and trigger again the sampling process
If you could share both results (RP2040 zero communication and working memory card communication) it would be really helpful to debug your issue.

dangiu, 2022-06-07:
Ok keep me updated 👍

sceic, 2022-06-09:
[image]

I moved the RP2020 zero into the same memory card case as the author, and the integrated size is very good. I have a feeling that the original problem might be related to the MCR file that was originally generated, but it's working fine now.

dangiu, 2022-06-09:
> 

Yes it's normal. Because firmware for these chips are statically allocated, meaning that they get always loaded in the exact same address. In this case the program does not overwrite the part of flash memory containing the filesystem with the MEMCARD.MCR file.
If you want to completely wipe the flash memory the Raspberry Foundation provides a special uf2 file called something like flash_nuke. It's near the bottom of this page: https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html

Anyway the initial MEMCARD.MCR file shown on the first boot is actually just a placeholder (if you look it's actually 0 byte in size). That's why no savefile was popping up in your bios. In the repo (under "docs/images") I provided 2 memory card images you can use to test.

dangiu, 2022-06-09:
> very good. I have a feeling that the original problem might be related to the MCR file that was originally generated, but it's working fine

That tight integration looks great dude!
I will now close the issue since the problem has been solved.

commit 2681a1ee (2022-06-09): Remove placeholder MEMCARD.MCR creation, fixes #3
closed 2022-06-09 by dangiu