dangiu/PicoMemcard #9: Pico Pi W

Androxilogin, 2022-06-30:
Wondering if this project will be supporting this wireless feature implemented in these new boards. That would be awesome!

Anarky, 2022-07-21:
Another possibility: the Pico W uses the CYW43439, which supports Bluetooth, [though it's not enabled yet](https://www.raspberrypi.com/news/raspberry-pi-pico-w-your-6-iot-platform/).
[A PS3 controller uses Bluetooth 2.0 + EDR](https://github.com/felis/USB_Host_Shield_2.0/wiki/PS3-Information#Hardware), and the [CYW43439 supports it](https://www.infineon.com/cms/en/product/wireless-connectivity/airoc-wi-fi-plus-bluetooth-combos/cyw43439/).

It just discovered [the BlueRetro project](https://github.com/darthcloud/BlueRetro) which uses an ESP32 to connect various controllers to various consoles.

What would be crazy would be to have a memory card **and** controller adapter in a single board, but I'm not sure if the Pico can handle it.

Androxilogin, 2022-07-23:
> It just discovered [the BlueRetro project](https://github.com/darthcloud/BlueRetro) which uses an ESP32 to connect various controllers to various consoles.

I wanted to do that project soo bad for a while since I don't have a GameCube Controller. I just modded both of my GameCubes with a Pico AND I used my ESP32 for my PS4. That would definitely be cool to see the new Picos compatible with that project.

dangiu, 2022-07-24:
Definetly something worth looking into

closed 2022-08-23 by Androxilogin