dangiu/PicoMemcard #14: Randomly disconnecting

Androxilogin, 2022-07-17:
I noticed this happening during the Saga Frontier load menu. It will read the data then disappear. If you choose your save quickly you can get it to load. I've tried it on two Phat models with the same result. https://imgur.com/a/tmWFf1v

dangiu, 2022-07-18:
I'll look into it, thanks for reporting!

Androxilogin, 2022-07-19:
Sure! Just happened to be the first game I tried it with. Haven't tried saving with that particular game since I was wondering if it was maybe my wiring and moved on to other titles to check.

dangiu, 2022-07-20:
Is the game Sega Frontier SCUS-94230 (NTSC version)? I'm a bit busy in these days but I'll definetly check it out... I have already a couple of ideas on what might be going on.

Androxilogin, 2022-07-23:
That's the one. I also have been taking inventory of a few other things that I noticed. I'm not sure if you have a regular memory card or not to go against but what I've noticed so far:

*If you plug the memory card in while the console is powered on it will reboot. 
- You're probably aware of this one but it is inconvenient. Perhaps nothing can even be done about it.

*While in the memory card menu in the bios it acts abnormally.
- With a regular memory card, once you copy some data to it it will just stay on the screen. I was reading your background on [how this project came to be](https://dangiu.github.io/2022/05/13/picomemcard.html) and noticed that the way it's writing to the ram is probably the cause. But anyway, figure it should still be mentioned. Copying data to it causes a few seconds pause and then everything is loaded in again. Which brings me on to my last report.

When you're in the memory card bios screen with a typical memory card and you accidently delete some data, there is a data recovery that can be done to salvage that save that was just deleted by pressing L1, L2, R1, R2 all at once a few times and it will load in each save one by one with the save showing back up in the list. With the Pico, this is no longer an option. 

And no problem. It's your project, I just think it's cool. As a kid I was always fascinated by the look and thought it was crazy it was capable of holding data. Especially animated tiles that held data! That was something else back in the day. 

I've also tested this with the MemCarduino project and it works as expected. Although unnecessary as it is. One last thing I was wondering about was why it had to use MEMCARD.MCR specifically as the name. Guessing it calls for that name specifically while searching for data? Is it something you plan on changing in the future?

dangiu, 2022-07-24:
Lots of things worth discussing.
Regarding the abnormality in the BIOS is exactly what you mentioned. The syncing process between ram and flash causes the problem. Unfortunately that cannot be avoided when using Pico internal storage. A mitigation I'm planning to introduce is lengthen the delay between sync (e.g. to 10 sec) and add an option to sync "manually" using a specific controller input (also blinking the led while the sync is in progress).

Regarding the data recovery I didn't know that there was such function! What kind of PSX model and BIOS version do you have? I need to check the code on how block delation is processed by PicoMemcard, it shouldn't be too hard to fix.

Regarding the name, yes it is checked in the code. I placed it initially and never improved on it... Can easily be changed :)

Androxilogin, 2022-07-26:
I actually have a SCPH-1001, a SCPH-7501 and a SCPH-101 I've been testing with. As for the bios versions, I'm not exactly sure. I've recently been trying to get things cleaned up a bit for new and incoming projects so ended up throwing them in the closet. I have another SCPH-1001 coming in a couple days to restore so I'll be digging out some old parts consoles, I'm sure. So far they've all behaved the same way consistent with all I've tested but I'll keep you posted if I stumble onto anything interesting or odd.

Androxilogin, 2022-07-29:
Update: I am not having problems with Saga Frontier anymore. It must have been the way the case I made previously seated the pins. I've also tested this on two different PS2 models and neither would work with them.
[image]
[image]

dangiu, 2022-08-02:
Awesome job on the PCB case, is the design publicly available?
Everything mentioned in this issue should be fixed in the new release.

closed 2022-08-02 by dangiu

Androxilogin, 2022-08-03:
You've sure been busy with this. I have not released the design publicly. The custom board was fitted very specifically and I may change the design again slightly. The MicroSD is a bit overkill for me personally but the 2040-Zero would allow for a smaller footprint. Curious to know, before adding the flash file, the device actually has somewhere around 200mb? Is this impossible to add a push button switch to the pi and enable different cards via flash if it were written differently? Also, do you plan on adding an LCD in the future? I'm not even sure if that's possible with these modules but it would be cool to see!

dangiu, 2022-08-03:
The Pico only has about 2MB of flash, the problem about supporting multiple cards (beside the limited space which would allow only a few) is having an intuitive way of transfering them. The solution right now is to use USB mode and an in-RAM disk as a sort of bridge between OS and pico's flash memory. Theoretically, using a button to select where the RAM disk maps to flash should be possible.
Regarding the LCD somebody already introduced support for a small OLED monitor during development, however this feature is not yet ready for release as introducing it would prevent users without one from being able to use PicoMemcard.

Androxilogin, 2022-08-04:
The drag and drop convenience is definitely a winner for me. As for the OLED, that's awesome! I don't know why I said LCD. This project seems it might surpass the MemCard Pro very quickly. Someone also shared this board idea from a similar project they found on the psx.dev discord.
[image]
[image]