dangiu/PicoMemcard #17: Where to find SD Card Modules

Androxilogin, 2022-08-08:
I meant to fit some cases for the variations of files offered here but I'm running into some problems. I bought two different kinds of boards and they each have their own set of issues. The **[Adafruit module](https://www.adafruit.com/?q=Adafruit+Micro+SD+SPI+or+SDIO+Card+Breakout+Board+-+3V&sort=BestMatch)** is well built but also a bit thick & pricey. While fitting with the basic Pico Pi (longer version) it seems like it may be impossible to fit in the case comfortably. The board itself is about 1.76mm without the card slot. About 3.70mm (height wise) altogether and 23mm in length. The other board is around 3.50mm thick but with a little over a whopping 42mm in length. 

What I'm looking for is what board you would generally recommend so I that if I were to share a case here that everyone would have a perfect fit. I'm also a bit confused with the **[wiring of the board I got](https://imgur.com/BZVWM92)** if this is what were to be recommended. Or at least for testing. 

I have a few of the 2040-Zeros on the way but it seems they were outsourced from China so it's going to take a little bit longer. I plan on making a case for both models, small and large with and without the SD module. Looking forward to shortening up the case to at least the regular size of an official memory card when my Zeros get here.

Raphfriend, 2022-08-09:
You can wire directly. I din't find any documentation, so i just did the test.
That's mine:
<p align="center">
  [image]
 <br>
  <b></b><br>
<br>

JAOT, 2022-08-09:
Heh.
I don't really care for pretty devices, as long as they work. I converted mine to a Pico

Memcard+ with an SD Connector I had that wasn't going to be used.
Had to connect the 3.3v and ground pins, though, as without them, the computer saw a drive but was unable to mount it.
[image]

Here's how it looks:
[image]

It's picky regarding the memory cards used, though, but I'm sure that is because of a mix of the adapters used and cards.
I'm currently using a 128MB card, and for the PSX that is unlimited saves! :D

dangiu, 2022-08-10:
For SD card modules I would reccomend looking up "microSD SPI" on a website like Aliexpress.
Regarding the connection of your module you can look it up on https://en.wikipedia.org/wiki/SD_card#Transfer_modes
In particular the Pico supports the card in SPI bus mode, so those are the pinout that you should use (the table gives you the conversion for your pinout to the one used by the wiring diagram).

@Raphfriend never seen an SD card wired directly, pretty cool!

Androxilogin, 2022-08-10:
From that wiki I found:

MOSI:D1
SCK:CLK
MISO:D0

So I'm guessing SS is CS? 
Also, 3v (Out) and GND should be connected? I've been trying with a different SPI that I got off of Aliexpress for arduino **[similar to this](https://imgur.com/fuGddas)**. I attached the VCC to 3v3 out and GND to GND, SS to CS but seems it's not working. Tried two different cards. Should it be formatted to FAT32? I've tried FAT & FAT32 but it hasn't shown up. Just an unhighlighted drive. Ejecting/inserting shows no difference. I've used these modules in other projects in the past and have came across a few that were defective. I have yet to test the one I've pictured above. Figured I'd ask first before I burnt something up.

dangiu, 2022-08-11:
Yes SS is CS and both 3v3 and GND must be connected. Your connection seems to be correct! Yes FAT32 file system should be used.
The picture you have linked is the exact same adapter that I'm using.
Let me know how it goes :)

Androxilogin, 2022-08-17:
That might explain it. I probably waited a total of around 10 seconds at most each try. I've since disconnected the wires for the current Pico I've been working with. Been waiting on my mini Pis to do some more testing. Figured they'd have been here a lot sooner. Ah, well. Hopefully I'll be back at it by tomorrow.

Androxilogin, 2022-08-18:
I hate to double post so here is my previous entry as well as my update:
*That might explain it. I probably waited a total of around 10 seconds at most each try. I've since disconnected the wires for the current Pico I've been working with. Been waiting on my mini Pis to do some more testing. Figured they'd have been here a lot sooner. Ah, well. Hopefully I'll be back at it by tomorrow.*

So yesterday I came up with a case and put it all together but I'm still having issues. I've tried three different SD cards formatted to FAT32, waited well over 30 seconds, I don't know what the deal is. When plugged into the console it blinks red. Supposing this is an indicator for it not being detected. Attached to the PC, it doesn't pass it through, still just a grey transparent drive showing up. I was too tired to comment here last night, here is what I came up with for the Adafruit and Pi-Zero board combo.

[image]

[image]

[image]

[image]

[image]

[image]

dangiu, 2022-08-18:
Very cool case design, one suggestion I have for troubleshooting: have you tried setting up the sd card (moving a 0.MCR memory card image) using a normal sd card adapter and then plugging it into the RP2040 and psx?
I'll try to recreate the same situation as soon as I have a bit of time. It might be possible that a bug of some sort slipped into the release and I haven't tested it properly.
If anybody else managed to get PicoMemcard+ working on an RP2040-Zero please post a comment.

Androxilogin, 2022-08-18:
Thanks- I forgot to mention that I had to snap the reset button off my board to have it fit correctly. I had a feeling it might be a problem as soon as I hit print but I let it go just to see. I'm fine with that but I should probably correct it. 

Yes, I have tried that. And I reformatted the card(s) and made it the only data that was on them just to be sure. I'll post a diagram of my pinout based on what you have here hopefully later today just to be sure it's 100% correctly wired up. Has anyone else here used the Adafruit board? I'd assume it should be a great candidate.

qxc86, 2022-08-20:
> If anybody else managed to get PicoMemcard+ working on an RP2040-Zero please post a comment.

Tested just now on the precompiled release image with no issues. I'm also using an adafruit breakout board like yours. Wired as follows:

```
MOSI > SI
MISO > SO
SCK > CLK 
SS > CS 
 ```
Which should be the same way as above. The only oddity I ran into was, I had some trouble until I wired both the PS1 power line and the breakout card via the 3.3V line on the RP2040 Zero, contrary to the README picture. Using an off brand 32GB card.

Androxilogin, 2022-08-20:
> > If anybody else managed to get PicoMemcard+ working on an RP2040-Zero please post a comment.
> 
> Tested just now on the precompiled release image with no issues. I'm also using an adafruit breakout board like yours. Wired as follows:
> 
> ```
> MOSI > SI
> MISO > SO
> SCK > CLK 
> SS > CS 
> ```
> 
> Which should be the same way as mentioned above. The only oddity I ran into was, I had some trouble until I wired both the PS1 power line and the breakout card via the 3.3V line on the RP2040 Zero, contrary to the README picture. Using an off brand 32GB card.

Holy crap, it freakin' works. My MOSI was connected to D1. Just grabbed the soldering iron real quick and moved the wire over to SI and Bam. Worked right away. Thank you soo much! Now I can actually get to messing around with this thing. I did have my 3.3v wire split off as you do at first but later switched it to the 5v. Seems to work fine here and the cards seem to switch super fast!

qxc86, 2022-08-20:
I would be a little worried about having the SD module wired to the 5V line if you ever connect via USB, but that's how mine was for a while and I didn't fry my card. Presumably the 5V pad is outputting 5V when USB is connected, but I haven't actually taken a multimeter to it. I could have also had a bad connection before I moved them, it's difficult to get two wires on the same pad lol. YMMV. 

Where did you get those memory port pads? The whole setup looks super good!

Androxilogin, 2022-08-20:
I was worried about that initially. My 5v line is connected to the memory card port connector to avoid any overdraw. While plugged in, the USB port is inaccessible to avoid any accidents. The boards were drawn up in KiCad. I had to wait around a month to get my initial model design finished since I was building around the board. A bunch of prototypes later I have two variations done. One for the original RPI2040 without a Micro SD and now this one that I can match up with different modules I test. Now I'm going to go off into a ramble outside of anything you asked, but it has to do with the topic of this title. 

* **SD Slot:** *
Most common seem to be the cheaper AliExpress models so I should order some of those to get fitted. I definitely want to go for the shorter boards to make it seem more authentic. 24mm X 42mm is what the page says but that could differ from the actual product. I'm especially confused seeing the size difference of the two pictured claiming to be the same size:

[image]

* **Case** *
I did mess around with the aesthetic markings of an original memory card such as the space for the labels and the triangle at the top but it came out as bad as I thought it would. It could be accomplished if printed differently which would take away a lot of the strength and open up the possibility of pieces snapping off inside console ports so I discarded that idea.

qxc86, 2022-08-21:
Haha yeah the measurements on the second one look a lil suspect. 

Hoping you're planning to post the board and 3d printing info publicly at some point. Would be very interested in making one, albeit I'd probably want to sub in the QTPY 2040 instead of the Zero. The Zero is the largest "very small" 2040 board on the market so if it fits, anything else should. 

How is the durability of the internal bracing? It feels like USB-C takes a lot of force to connect and disconnect, at least relative to how big these things are.

Androxilogin, 2022-08-21:
Being that the port connector boards are extremely inherent to the design I'm not sure how I want to go about it yet. I figure I'll cross that bridge when I come to it. I'm not too keen on moving into the whole USB Type-C aspect of it. I stubbornly still feel that the previously released generation is sufficient being that most of my devices use them but I did buy some cheap converters to quell my ignorance of their usefulness. 

If there is a file released here for the QTPY 2040, I'll most likely make a design for that as well. The durability is excellent. 1mm walls are surprisingly efficient with small components. There's also room to spare in making the walls larger but I don't see any need to. I did mention previously in this thread about some really bad filament I was using to prototype that was hilariously terrible. I discarded it for accuracy but it was an interesting experiment. Very light pressure and I was snapping pieces off everywhere. It was a spool that came with my printer which I completely forgot about so it was worth a test. Everything came out pee colored and delicate as a waffle chip. 

The USB-C connector actually sits about a mm behind a wall so part of the case can slide forward past it as it's being locked into place. The zero dips into the bottom .5mm to make room for the underside components and rests on top of four 1mm high pillars. Then walls surround three sides for most of the length to give it a really robust positioning. I went back and forth on a forth wall coming to the conclusion that it didn't need it and could potentially over complicate things. The inner walls came up from the bottom of the board with enough spacing as not to pinch the wires even if they happen to overlap. At some points it was a bit rough fitting everything into such a small height. 

I've also ordered some of those shorter SD modules just after my last comment here. I have one of the longer version I was using for other projects, just not sure if I want to build something around it due to the size. So hopefully the size in the picture above is a lie for these new ones. My micrometer says this other board is correct.

closed 2022-08-31 by dangiu