dangiu/PicoMemcard #18: something wrong with current PicoMemcard+.uf2 file?

copywizard, 2022-08-15:
Hello after some testing i first thought my pico was broken or my soldering skills where to blame but somehow if i download the current PicoMemcard+.UF2 file and try to flash it to my PICO it closes after copy and does not pop up the PICO MC usb storage.
But when i use the PicoMemcard.UF2 file it works perfectly!

so my conclusion is there is something wrong with the +.UF2 file

qxc86, 2022-08-16:
Can confirm this behavior, my assumption is that reading and writing was intended to be done with the microSD card rather than the USB port with pmc+ . Unsure if this is a correct assumption or not.

copywizard, 2022-08-16:
@qxc86 do you mean if i flash the + image i just have to wait 30 sec+ for the usb storage to show up?

dangiu, 2022-08-16:
Have you verified to have a working SD card (and SD card module) connected to your pico and formatted using FAT32 filesystem? Otherwise it will not show up.
When using PicoMemcard+ the Pico simply acts as a passthrough for the SD card but reading and writing is somewhat slow.

copywizard, 2022-08-16:
ok thats something i did not know i had hoped it would work without the sd card then i was just mis informed this can be closed.

p.s. i will get a sd card SPI module asap

Raphfriend, 2022-08-16:
I recommend that you use another SD reader because the RP2040 is quite slow in terms of transfer rate. As @dangiu  mentioned, it's probably expected behavior.

copywizard, 2022-08-16:
Sorry I did not mention this but I,am using a full size RPI pico and a daughter bord from dangiu.

I soldered on the RPI pico and it works fine used it all day with the normal version of picomemcard but when I put on the + version it just does not work.

But I have ordered some micro SD card spi modules and it seems the daughter bord from dangiu already have the proper pinout for it on the back so when it comes in I will solder that on and hope it will work with the + versions then?

Raphfriend, 2022-08-16:
@copywizard If you look at [this topic](https://github.com/dangiu/PicoMemcard/issues/17#issuecomment-1209271257), you'll notice that I'm using direct wiring on an SD to microSD adaptor, and it's working fine. However, when I try to read on my PC, it is quite slow, either in terms of being recognized by Windows or transfer rate (about 88Kb/s).

I think any microSD addon should work.

copywizard, 2022-08-16:
@Raphfriend thank you I have seen this topic before but I hope to do a more clean job right now the dangiu daughter bord and pico work fine I will post the final results when I get the SD card reader in and working hopefully the write speeds will be a bit better then mentioned?

dangiu, 2022-08-16:
> @Raphfriend thank you I have seen this topic before but I hope to do a more clean job right now the dangiu daughter bord and pico work fine I will post the final results when I get the SD card reader in and working hopefully the write speeds will be a bit better then mentioned?

Yes, if the pinout of the SD expansion board is the same it will work right away... I've based my design on modules commonly found on Aliexpress (e.g. https://it.aliexpress.com/item/32877331401.html)

Raphfriend, 2022-08-16:
We don't need the regulator because the pico (even ZERO) already has 3V3, so what's the intent of the second IC?

copywizard, 2022-08-16:
@dangiu I have ordered that precise one and the smaller version here that somehow has another pinout.

https://a.aliexpress.com/_v01seV

qxc86, 2022-08-16:
> @qxc86 do you mean if i flash the + image i just have to wait 30 sec+ for the usb storage to show up?
> 
> Because if i do so i see a new usb device pop up but it is unusable/inaccessible

Less than 30 seconds, it's not quite as fast as the onboard PMC flash as far as showing up. Testing a few times just now it took 8-10 seconds. It did take around 30 seconds after clicking the disk icon in linux for it to finally load the contents however, long enough to think that it was not going to work. As mentioned, using the USB to manage files on the microSD card is very sub optimal. You're much better off just attaching the SD card directly to the computer via adapter.