dangiu/PicoMemcard #20: LED Not working on RP2040-Zero when building from source

qxc86, 2022-08-19:
I can not get the LED to function as expected on the Waveshare RP2040 Zero when I build the code myself. I currently have the board wired up only to the PS1 pins, no SD card pins wired. The expected behavior is a flashing red LED when connected to the PS1 indicating it can't read the SD card image. I've confirmed that the current release image available does work as intended, and to my knowledge there are no changes between that and the current release branch. 

I'm building from the `smc+/release` branch, with the only change being the following:

```
$ git diff
diff --git a/inc/config.h b/inc/config.h
index 218120d..f2780f0 100644
--- a/inc/config.h
+++ b/inc/config.h
@@ -10,8 +10,8 @@
 #define MC_RECONNECT_TIME      1000                            // time (in ms) the memory card stays disconnected when simulating reconnection
 
 /* Board targeted by build */
-#define PICO
-//#define RP2040ZERO
+//#define PICO
+#define RP2040ZERO
 
 /* PSX Interface Pinout */
 #ifdef PICO
```
Which should set the target board appropriately. My build process is:

```
export PICO_SDK_PATH=~/pico/pico-sdk/
mkdir build && cd build
cmake ..
cmake --build . --target all -j 10
```
Building on Ubuntu. No warnings except for a couple at the very end, all part of the SD submodule.

dangiu, 2022-08-20:
I bet the problem is that compilation requires a fix for pico-sdk (regarding multicore lockout functionality). I only mentioned in the commit messages but it got buried in the history.
Here is the relevant pull request:
https://github.com/raspberrypi/pico-sdk/pull/936

qxc86, 2022-08-20:
Ah yes, I remember reading about that but had assumed it was merged into the main branch already.

I changed my pico-sdk folder over to `development` branch, confirmed that the commit in question was present in git log and confirmed `PICO_SDK_PATH` was pointed to that folder, but my built binary still did not work. Went back and forth between that and the precompiled to make sure I didn't break a wire. I'm assuming this is a failure in me knowing how to correctly pull in those added changes to the SDK now.

dangiu, 2022-08-22:
Is there any possible way for you to debug the program and see where it gets stuck? That would really be helpful

qxc86, 2022-08-23:
Is it possible to do that without using the debug pins on the RP2040? The QTPY RP2040 and RP2040 Zero don't have debug pads on the board. I tried soldering wires directly to the pins coming off of the RP2040 but that's beyond my capabilities. My basic understanding would assume since USB is used for presenting as a USB device for the PC, it can't be used for debug?

dangiu, 2022-08-25:
Exactly, unless some heavy code modifications are present but that would defeat the purpose of debugging the current code :/

dangiu, 2022-08-31:
Are you still having the same problem? Can you try with a different machine?

qxc86, 2022-09-06:
Just got around to trying the new version this weekend. Same issue as before, mostly. 

To try and isolate the issue I got a fresh image of raspberry pi os and loaded it onto my pi 400 so I could start from scratch, including the env. 

```sudo apt install cmake gcc-arm-none-eabi libnewlib-arm-none-eabi libstdc++-arm-none-eabi-newlib
cd Documents/
git clone https://github.com/dangiu/PicoMemcard.git
git clone https://github.com/raspberrypi/pico-sdk.git
cd pico-sdk/
git submodule update --init
git checkout develop
cd ../PicoMemcard/
git checkout pmc+/release
<edit inc/config.h to comment out #define PICO and uncomment #define RP2040ZERO >
git clone https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico.git
mkdir build && cd build
export PICO_SDK_PATH=/home/pi/Documents/pico-sdk/
cmake .. && cmake --build . --target all -j 5```

I also tried with `pico-sdk` checked out to `fe7849d` and the fatFS module to `4f09f02` with no change. Pardon any typos, I had to copy this all over by hand because I didn't want to log in via the pi. I did just go back and retype them into the pi with the same results however. 

Why didn't I use submodule update for the PicoMemcard repo? Because it fails with access permissions. :shrug: 

Hopefully something in there is obviously wrong, or you're able to replicate on your end.

realJoshByrnes, 2023-03-19:
https://github.com/dangiu/PicoMemcard/pull/44 should fix your access permission issue. Can you try pmc+/development branch?