dangiu/PicoMemcard #22: Onboard Hardware Buttons

Androxilogin, 2022-08-27:
In reference to the limitations of the PS2 controllers being on a different bus, I was wondering if it would be possible to use the onboard Boot & Reset buttons optionally to switch cards. 

Also, on the idea of implementing PS2 Memory Card emulation in the future I was wondering if you also had plans to add the same abilities as **[MC2SIO](https://heldergametech.com/shop/ps2/mc2sio-microsd-iso-loader/)** to load ISOs through HDL. I know it's a pile of questions to throw out there but this would be revolutionary. Say- if the Boot and Reset to switch cards back and forth is possible, something like hold Boot for five seconds to switch between PS1 and PS2 card emulation modes verified by blinks or colors.  Hold reset for five seconds to enter PS2 ISO emulation mode. That would be soo handy with McBoot. 

Perhaps tap the Boot button twice to create a new PS1 card, Reset twice to create a new PS2 memory card. Both automatically switching to the new card and verified with a blinks; color hold while creating. 

Again, just throwing out ideas I've been curious about while ignorant to the limitations. Not saying, "oh, you should totally do this!" Also wanted to **[share this crumby video](https://www.youtube.com/watch?v=8dUKlE79xmk)** I made the other day showing off that card I put together with the Adafruit module as well as the button functions (before the last update).

qxc86, 2022-08-27:
I don't think the actual GPIO calls are in the code but that did get implemented as part of [this yet to be released product](https://twitter.com/angryhelder/status/1554542618939359232) to my knowledge. It's something I also want to put in whenever I'm done porting the code to other RP2040 boards. 

My limited, probably incorrect assumption about how MC2SIO/MX4ISO works is that it's communicating _directly_ to the PS2 by using a driver they wrote to interface with the SD card itself. The boards made for this don't have any ICs on them, [they're mostly passive adapters](https://www.psx-place.com/threads/mx4sio-sio2sd-sd-card-adapter-and-sd-driver-for-the-ps2-sio2-interface.29210/). I don't know if you could pass that through the RP2040 at all, let alone without additional latency issues or what not. 

Looking forward for public files for those cards in the video, by the way! I wanna build a couple up that aren't a bunch of parts hanging out of a torn up third party card lol

Androxilogin, 2022-08-28:
I'm pretty much assuming that some of these questions I had are not possible but figured I'd throw it out there for those who might know. 

As for my case files, I'm not real sure if there will be a public release or not. There are many reasons for this, including the fact that I've built these around custom PCBs I designed for them to fit exclusively. This was unintentional but seems I won't be turning back now on these current designs. I plan on reworking these boards based on the same sizes but with through-hole vias in the future to make soldering cleaner and easier. The thickness and width are not sized as the original and with such connections, small imperfections as low as 0.1 could cause it not to function whatsoever. I figure I'll decide once I get to the point of satisfaction with all variations I've been working on, where I want to go from there. While I do appreciate contributing to the community, I've had designs in the past taken, branded and mass produced for profit without my knowledge or consent so I'm very cautious of what I distribute. I'll work in many aspects of providing diagrams, bug reports and suggestions, along with getting the word out to build interest. If I knew more about the coding besides recognizing only what has already been written I'd assist with that. Outside of that, I prefer to keep my models independent.

I do **[offer a number](https://www.thingiverse.com/androxilogin/designs)** of simpler designs I've created that are useful for other consoles. **dangui** has made his own single boards and has planned a case for it in the future as well. I'd have built a case around their PCB to publically distribute if I had one.

dangiu, 2022-08-31:
@Androxilogin can you open a different single issue regarding only onboard hardware button? I'll label that as a feature and close this one since most of it's content is off topic, that way I can keep things more clean and organized following a logical thread.
Regarding your questions I will answer more in detail in the dedicated issue... The TLDR is that using the Boot switch has very heavy limitations and is not possible for most of the stuff I would like to do, using the Reset button is straight impossible since is tied directly to the pin that resets the MCU.

closed 2022-08-31 by dangiu

Androxilogin, 2022-08-31:
It's fine, I'll just leave it in your hands and see whatever develops.