dangiu/PicoMemcard #24: Image Swtitching Hardware Button

dangiu, 2022-09-02:
As mentioned by @Androxilogin since on PS2 memory card and controller bus are separate PicoMemcard cannot detect controller input therefore memory card switching is not functional.
Additional hardware buttons are required for memory card switching, which likely requires the availability of two GPIO inputs (for two buttons) and software support.

qxc86, 2022-09-06:
If I ever get builds working properly this was next on my list after extra board support.

g0ldenflash, 2022-09-12:
And maybe these buttons can do other things like creating a new mcr by long press or double click?

dangiu, 2022-09-13:
That's the idea

LobAnovich, 2023-01-21:
In this case, an OLED screen would also be useful, which would display the number of the active memory card

dangiu, 2023-01-23:
Yes, the plan is to get some sort of incremental build with components that can be optionally toggled on and off. This way people that want the chepest possible solution don't have to spend a lot of money, while others that want a full fledged version can have it

realJoshByrnes, 2023-03-19:
It's possible for us to poll the existing button on the Pico/Pico W. Not sure about the buttons on the Zero. I don't believe it's possible to set an interrupt, but I haven't investigated it.

dangiu, 2023-03-19:
I remember looking into that but it was not practical (or straight impossible if I remember correclty) mainly for two reasons:
- We are using both cores simultaneously
- It requires code running from ram which we barely have enough of
- (maybe it even requires suspending interrupts ???)

Don't take my answer for granted tho, I'm writing this from what I remember off the top of my head.
If it's possible it would be the ideal solution tho.

realJoshByrnes, 2023-03-19:
Using both cores simultaneously was an issue I had considered, It should be possible to poll when the MemCard is not active since we have interrupts on that.

That is the next issue; It does require suspending interrupts. (I didn't know that)

https://github.com/raspberrypi/pico-examples/blob/master/picoboard/button/button.c

einso, 2023-08-28:
Hi! Just wanted to check in on this: I recently built some PMC+ as my first experience with ordering parts/PCBs, soldering components etc. 
I noticed the buttons on both my units are not working, whether inserted in a PS1 or a PS2 (switching via controller works on the PS1 obviously). 
Wanted to know: are the buttons supposed to be working at this point? I see the issue is still open (and, understandably, hasn't been touched in a while), but the ReadMe makes it sound like it should be working.

dangiu, 2023-10-09:
Nope the buttons are not implemented yet, are there on the pcb just for future proofness