dangiu/PicoMemcard #25: After installing appropriate release to RP2040-Zero PicoMemcard doesn't appear as USB drive

AmenAnthem, 2022-09-04:
I did the installation according to the readme, but unfortunately my RP2040-Zero doesn't reppear as a USB drive, after dropping the PicoMemcard+RP2040-Zero.uf2 file onto it. It disappears (with Windows classic plugout sound effect), then the replugin sound effect plays, but the device is nowhere to be found. I tried it with all the releases already to no avail.
Any help is appreciated!

AmenAnthem, 2022-09-04:
After trying it multiple times for whatever reason it worked now.

closed 2022-09-04 by AmenAnthem

GarnetSunset, 2023-06-15:
Same exact issue here, the memory card even flashes green whe it's plugged in

dangiu, 2023-06-19:
I've seen this issue mentioned quite a few time... There seem to be no relibable way to reproduce it as it's kinda random. You can always load the files on the microSD with a separate adapter and then plug it into the pico and directly into your playstation (without doing the USB transfer thing)

GarnetSunset, 2023-06-21:
Turns out *I* was just installing the + version on a non plus rp2040, that was my fault, I thought it was related but it wasn't, would it be possible to include a picomemcard (non plus) version in the latest releases so people like me (idiots) don't get confused?

dangiu, 2023-06-27:
Yes, I'll update the page to provide a link to the latest release of both version :)