dangiu/PicoMemcard #26: Not enough disk space?

NovenTheHero, 2022-09-07:
I am trying to put the blank memory card file on to the pico and it says there is not enough free disk space. Super weird!
[image]

NovenTheHero, 2022-09-07:
[image]

Androxilogin, 2022-09-08:
You're overwriting the MEMCARD.MCD file, correct? Just delete it and throw on the new file.

NovenTheHero, 2022-09-08:
No, just installing the firmware, then dragging the memory card file on to the folder.

dangiu, 2022-09-08:
Sometimes windows automatically puts in garbage data to keep track of some stuff... I specifically enlarged the disk above 128KB to make sure this is not a problem but apparently it still seems to be!
Could you test something for me? Can you download the nuke_flash.utf2 from [this page](https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html#resetting-flash-memory) and flash it onto the raspberry to ensure that all the data gets wiped. Then reinstall PicoMemcard and check if the problem is still there

NovenTheHero, 2022-09-08:
I did that and I am getting the same results. This is super weird, and mildly frustrating.

dangiu, 2022-09-09:
You are on Windows 10 correct? Could you try to repeat the same procedure (nuking the Pico flash memory) and flash PicoMemcard from a different os? (maybe a live linux distro such as ubuntu). I'm sorry but I can't to much to debug the issue since it seems to be computer specific.
If anybody else is having this problem please report here your machine configuration.

NovenTheHero, 2022-09-09:
Super weird, I did the nuke, then reflashed it on my linux machine and I think it worked. I will test it out on real hardware later when I get home from work, but tentatively, it is good. At least no issues copying the memcard.

arkwise, 2023-06-02:
hmm weird.. ive tried Mac + linux and windows and the picoMC device shows up but there is no partition or anything,

I flashed the uf2 file fine and the picoMC drive shows up afterwards but there is no partition at all just the device. 

when you try to open it or drop file it just says please insert removable media (win) and Mac and linux just say there is no drive/partition available.

I then re-bootsel and dropped my mcr's into this and the uf2 last and all it flashes but no memory card images are available.

dangiu, 2023-06-08:
Which version did you install? You need the plus version for using the micro sd (and of course a micro sd extension and the micros sd itself)

arkwise, 2023-06-11:
> Which version did you install? You need the plus version for using the micro sd (and of course a micro sd extension and the micros sd itself)

the standard version I am not using a SD CARD for storage.

dangiu, 2023-06-12:
> Sometimes windows automatically puts in garbage data to keep track of some stuff... I specifically enlarged the disk above 128KB to make sure this is not a problem but apparently it still seems to [...]

Have you already tried to to this?

arkwise, 2023-07-26:
> Have you already tried to to this?

sorry for the late reply, i get an error 1407 on my mac, now i have about 10 of these unit and i get the same results and theyre official units and ive purchased them at different times over the past few years and function fine for other projects so i know theyre mint and ive tried the nuke flash and both version of pico meem to no avail.  the only difference is the nuke remounts the storage after i loaded nuke onto it but as soon as i try to load pico mem its back to square one.

im so keen on this for another project im working on so any help is appreciated mate :)

dangiu, 2023-10-09:
Can you try on a different system? Possibily some linux distribution, not on an apple machine