dangiu/PicoMemcard #28: blinking green light on psx

MrFilz, 2022-09-12:
I build my own picomemcard+ thanks to your project, but i can not get it to work on my playstation. I can write the pico+ firmware to the raspberry, and after a half minute or so the sd card folder opens. It is a 4GB micro sd fat32 formated with 0.MCR and 1.MCR on it. If i plug it into the playstation there is only a blinking green light and no card is recognized... Am i doing something wrong?

dangiu, 2022-09-13:
That's an issue I haven't seen before. So the SD card pops up when the Pico is attached via USB to the computer but does not work on the PSX and keeps blinking? The only thing that comes to my mind is that the microSD is not powered correctly when attached to the PSX, could you share a photo of the build? (in particular the various connections?)

MrFilz, 2022-09-13:
I used a cheap china clone mc for my build. I disconnected all pins from the rest of the board, and attached a pico on top. insulated the bottom of the pico with kapton tape and soldered the pico with its ground points to the board. So in the end every trace beneath the pins of the mc are now ground. This gave me a nice and solid connection. This was for the picomemcard. And it worked flawless after i wrote the firmware and disconnected it save from my laptop! A few weeks later i decided to try picomemcard+ and ordered basically the sd module from your github pictures. I desoldered the  headers and put it on top of the mc housing. Connected all with wires like this:
PIco 33 - ground SD module
Pico 40 - VCC SD module
Pico 21 - MOSO SD module
Pico 25 - MOSI SD module
Pico 24 - SCK SD module
Pico 22 - CS SD module

So the sd card is powered correct if connected to usb. 5V goes in, and the power regulator on the SD module provides 3,28V, according to my multimeter, for the sd card. But the Playstation is only providing 3,3V, so there are only 2,15V on the sd card after the regulator... but you used the same board on the pictures, so i wonder if i am missing something! I also triple checked everything, no shorts to ground on any pin besides ground, no shorts between the pins...

MrFilz, 2022-09-13:
[image]
[image]

MrFilz, 2022-09-13:
[image]

MrFilz, 2022-09-13:
[image]

MrFilz, 2022-09-13:
[image]

MrFilz, 2022-09-13:
I know it looks wild, but everything is connected like i mentioned it, and there is no short!

dangiu, 2022-09-14:
You didn't miss anything, looks like your build was well executed too! Which playstation model are you working with?
There is something that you could test.... If you have another Pico on hand you could: disconnect the power provided by the playstation, then connect the second Pico to USB and bridge the VBUS pins (with a jumper wire) in order to power the other Pico (PicoMemcard+) using USB power while connected to the PSX and see if everything works.

MrFilz, 2022-09-14:
Thanks for your kindness. I have a pal 5502 model with xstation ode installed. Interestingly enough the raspberry pi works flawless as a picomemcard. The issue begins if i try to turn it into a picomemecard+. So i thought the problem must be the sd module itself. And thanks to your guess with the power issue, i tried something else, because i do not have a second pico at hand. Pin 36 on the pico is 3,3V no matter what. It is 3,3V with USB and with the playstation. So i connected pin 36 of the pico with the power rail of the sd module, the right side of c3 on the sd module to be precise! And it worked! The picomemcard+ gives me a solid green, and everything works now! Maybe this helps somemone later! Thank you for this awesome project, and for your help!

MrFilz, 2022-09-14:
[image]

dangiu, 2022-09-15:
Awesome! I'm glad everything works! :)

closed 2022-09-15 by dangiu