dangiu/PicoMemcard #29: Picomemcard+ can't figure why I get a green blinking led

matvenu, 2022-09-23:
I updated my raspberry pi pico from a picomemcard to a picomemcard+ image. While On the picomemcard image, everything worked fine. When I updated it to the picomemcard+ , my pc recognizes the pi as a usb drive. On my psx though, no save files apprear or recognizes that I have inserted a memory card. I got the sample mcr file (which worked fine in the picomemcard) named as 0.mcr, as well the blank save file, named 1.mcr. 
I am using the exact micro sd module,as the one pictured on the github page. I solder cs to pin 22, sck to 24, mosi to 25, miso to 21, vcc to 39,gnd to 23 . I used both a 1 and 16 gb cards, formated to fat32 with 512 allocation space. 
What else should I do?

dangiu, 2022-09-25:
Can you try connecting VCC to pin 36 (3v3 out)? I used the same connection as you did and everything is working fine for me but some users reported having problems when powering the SD modules with VSYS (only when connected to the PSX which provides a lower voltage than USB) and through testing we figured out it was a power issue

matvenu, 2022-09-25:
I did as you suggested, but I still get a green blinking LED. Maybe I'll try with a 3v module, as the one I already have seems to work only in 5v

matvenu, 2022-09-27:
It was indeed my sd card module. Swapping it for a 3v3 one, the led stays green, as well my psx recognizes it as a proper memory card. It seems that there are variants of the modules that operate only in 5v, even though the sellers claim that function in lower voltages

dangiu, 2022-09-27:
I'm glad everything works :)