dangiu/PicoMemcard #30: Strange issue with Toy Story 2 Game

carmax91, 2022-09-28:
Hi,
I'm using the RP2040-Zero Version v1.0.2 with Toy Story 2 [ITA ver SLES-02407] and when i try to load the save from PicoMemcard+ the game fails to load. Aftyer various attemps the game succed to load and i can continue the game normally.
Seem the game use a strange way to manage the memory card because even if I try to switch the MC image with the controller game behave randomally glitching at the moment of save, like half MC loaded or corrupting randomally the image, owerwriting a save even of another game.

The MC works great with other games...

I've attached the save file to test, maybe someone can try and confirms this strange behavior...

P.S: I'm using the latest release of xstation if it matters.

[Test MC.zip](https://github.com/dangiu/PicoMemcard/files/9667263/Test.MC.zip)

dangiu, 2022-09-30:
I'll try to test it out as soon as I have time, thanks for the feedback!

whitezombie2000, 2022-11-27:
I can confirm this issue with PicoMemcard+ and a RP2040. I tested the Italian version and was sometimes getting failed to load when loading the save. I also tried the English NTSC version and it sometimes failed to load as well.

dangiu, 2022-11-27:
Sorry, I haven't had time to check it out yet. If you would like to do some testing try to see which sectors are being read/written by connecting a UART interface to PicoMemcard and seeing what gets printed out.

whitezombie2000, 2022-11-29:
Sorry, I don't have a UART device nor any knowledge on how to test it further.