dangiu/PicoMemcard #32: My PicoMemcard+ is not reading any images at all? Stuck blinking red

ilickmonkies, 2022-10-08:
Like the title says, I soldered it all up and I know everything with the SD card works because when I put the image for the RP2040 into it it automatically switches to the memory card through the USB-C port. I tried it with the correct version of FreePSXBoot and one I got from PCSXR, both named 0.MCR when I put it in. Do you think this is a software thing or something I did?

ilickmonkies, 2022-10-09:
The non-plus version works fine without the micro SD card module, I tried every format for the micro SD card just to see and that didnt work, so I will triple check the soldering on it, use a voltmeter and if that doesn't work I'm going to try out the DIY one of using the micro sd to SD converter

dangiu, 2022-10-10:
To recap: the Plus version SD works when plugged into the computer (because it pops up and you can transfer the image) but when plugged into the PSX the led blinks red correct?
In that case I believe it's a power issue... Some users have reported that their SD modules do not work correctly when plugged into the PSX (because of the low voltage). One of the solution was soldering the VCC input on the microSD card module to the 3.3V out of the Pico/RP2040-Zero).

ilickmonkies, 2022-10-21:
Sorry for a late reply, I tried this out. SD Module VC in the 3V3 pin hole for the RP2040 and now when I connect it to my PC it does not show up as a USB device. Quadruple checked all the wiring again and it all checks out so I have no idea what's going on. I also seem to be using the same module that you do

dangiu, 2022-10-22:
Ok let's try a new idea... First of all, which playstation model are you using? Second, do you have another Pico on hand?
If the answer is yes, you could try to:
- revert back the wiring so that at least PicoMemcard shows up in the PC (therefore showing that is powered correctly)
- tape away (or disconnect) the pin providing 3.3V from the Playstation
- power PicoMemcard using the second Pico plugged into USB (therefore power goes from USB -> Pico -> PicoMemcard)
At that point I expect it to show up when plugged into a playstation which would indicate that the problem is indeed related to the power the playstation provides.

Zombar3, 2023-05-29:
Hi. I soldered pico zero with sd card, flashed it with PicoMemcard+ and the sd card works on PC, but doesn't work on PS1, tried to flash PicoMemcard not + and also doesn't work on PS1, red LED flashes shortly.

dangiu, 2023-05-30:
You say that you have a "pico zero", do you mean an RP2040-Zero? In that case did you flash the PicoMemcard+.uf2 or the PicoMemcard+RP2040-Zero.uf2?

dangiu, 2023-06-01:
Ok thanks for the info. Since you are able to read the microSD card from PC but it fails when plugged into the PS a possible source of error could be a faulty memory card image. Where did you get your image? What file format is it_

Zombar3, 2023-06-01:
I use MemcardRex for save memory card image. Images in "docs/images" have size bigger then size on pico without sd . Red LED flashes 2 times shortly every time with sd or without.

Zombar3, 2023-06-01:
1. Windows 10
2. I use empty image from MemcardRex
3. Rename image with 0.mcr
4. Now i flash picomemcard blinks red LED, picomemcard+ blinks red LED
5. I try different SD cards 4, 16, 8 GB FAT32 formated

dangiu, 2023-06-01:
Everything seems fine to me... Could you try with a non-empty image as well? On some batches of RP2040-Zero the LED is inverted (green-red) it may be unlikely (but still possible) that everything is working fine but the board still shows the red led.

Zombar3, 2023-06-02:
First blinking green, then constantly blinking red

closed 2023-06-08 by dangiu