dangiu/PicoMemcard #35: Controller not working when i plug the picomem+

poyigiygas, 2022-11-30:
I manage to solve the ploblem with the blinking light in the pico and the pc recognoice the sd when i plug the usb in the pico, but it still disconnect the controller, even in the begining before i added the sd module. I put everything in the adapter pcb and it seems the connections are fine.

dangiu, 2022-12-01:
That is very strange, I've never seen that issue. Beside the controller not working, does the psx detect PicoMemcard as a valid memory card? What about if you plug PicoMemcard in the second slot and the controller in the first slot?

poyigiygas, 2022-12-02:
Nope even the status led says all right the ps1 dont recognice the picomemcard+.
When i plug in the second slot Does the same think, even sometimes reboot the console.
If it helps my console has been modded whith xstation and external psu.

dangiu, 2022-12-02:
The fact that it's modded shouldn't make a difference. It may be a power issue since the PC recognizes it. Could you try to do the following:
- tape off the contact that provides power from the PSX to PicoMemcard
- power PicoMemcard using a USB phone charger (do  not use a USB port of a PC otherwise it will enter in transfer mode)
- connect it to the PSX and see if it gets detected

poyigiygas, 2022-12-03:
I give up not working, anyway im going to buy the rp2040 zero and smaller sd adaptor and make it more tidy.
Thanks.

closed 2022-12-03 by poyigiygas