dangiu/PicoMemcard #36: Blinking LED (Pico)

BRIK1111, 2022-12-08:
I built the original PicoMemcard and everything was fine. I upgraded to the plus using the same SD card module as the guide and receive the blinking LED 99% of the time. Strangely, there have been one or two times when the test memory card file provided does appear.

I uploaded the new uf2 file, the SD card appears through the Pico on my computer for file transfer - so I'm confident the wiring is correct and that the SD card expansion module works. I've also measured the voltage on the module as it's plugged into the PlayStation. 3.5v more or less on the dot.

dangiu, 2022-12-12:
Since the SD card pops up when connected to the PC we know that the SD module and microSD card are both functional. To test whether this is a power issue I would suggest trying the following:
- Make sure that when connected to the PC the SD card pops up
- Tape off the contact that provides power from the PSX to PicoMemcard
- Power PicoMemcard using a USB phone charger (do not use a USB port of a PC otherwise it will enter in transfer mode)
- Connect it to the PSX and see if it gets detected when powered this way

Make sure to only disconnect the power pin and not the ground pin (to the PSX) otherwise they will not be able to communicate properly.
Btw which PSX model do you have?
Sorry for the late reply, let me know how it goes :)

BRIK1111, 2022-12-14:
Yep, taping off the 3.3v and plugging in a charger worked. Does this indicate that the module doesn't work as well on 3.3v as advertised? 

The PS1 models I've tried are the SCPH-102 and the SCPH-7502 (both PAL).

dangiu, 2022-12-14:
Yup, that is likely the reason why. Are you able to try with another SD module? Btw could you post a picture of your current one just for reference?

BRIK1111, 2022-12-14:
I've ordered another, this time from a reputable seller as opposed to eBay. I'll report back if it works. 

Here is a photo of the original:
[image]

dangiu, 2022-12-16:
Aesthetically is very similar to the one I was using, I'll see if I still have it and compare the part number of the power regulator.

BRIK1111, 2022-12-31:
I tried another that looked nearly identical (it's just missing the text that says "HW-125") and had the same issue! But again, taping off the 3v contact and plugging in a 5v power supply worked. Pretty disappointing since this was also advertised as specifically being compatible with 3v.

[image]

So, I bought this since it has a dedicated contact point for 3v power: https://www.jaycar.com.au/arduino-compatible-sd-card-interface-module/p/XC4386?pos=1&queryId=1f13511ba25c52c03adf5557b7537b72&sort=relevance

And it works! It's a shame that it's so big since it used a full-size SD card, but I'll take it as a win. 

[image]

dangiu, 2023-01-03:
I'm glad that at least now it works, I've changed the title of the issue so that it may be easier to find for other people that have the same problem. Happy new year!

closed 2023-01-03 by dangiu

dazzaXx, 2024-03-03:
Hi, I just wanted to add onto this issue here as I also had the exact same problem. Instead of looking for another module, I learn't that if you bridge pin 2 and 3 on the regulator, it bypasses it so you can use 3v. Simple fix. All I did to bridge them was put a blob of solder in between them.

[image]