dangiu/PicoMemcard #37: pi2040-zero can't find micro sd

MrPalloncini, 2022-12-31:
Hi, i soldered and troubleshooted everyting  by following the readme, howewer, when i connect the rp2040-0 the the pc after i flasched it, the sd won't appear and it flash red. Tryied also with a raspberry pi pico and it works, is there any problem with the fw maybe?

Thank you for thia awesome project

dangiu, 2023-01-03:
No, the firmware has been tested by many people, normally when the SD card does not appear is related either to a power problem with the module or due to the card not being formatted in FAT32. Since you said that it works fine on the pico I think it may be due to a power problem, make sure to double check your wiring and the voltage the module requires... It could be that the pico and rp2040-zero are outputting slightly different voltages.
Also check out issue #36 which discusses power related issue with modules.

seanchapman, 2023-01-19:
I'm having this issue too, as well as the inverted red and green LEDs (on a recent RP2040). Firmware v1.0.0 works but v1.0.2 does not.

dangiu, 2023-01-20:
About the LED problem, have you looked into issue #38? It seems like you are not the only one having it.
Regarding the disk instead I have no clue what may be going on? Can you provide some more info (what works on which platform (OS) and what does not with the different firmware versions)? By chance are you able to debug the program and step through the code?

seanchapman, 2023-01-21:
I've just tested the inverted LED build and actually that's working great for me now! LED is the correct colour, saves are loading on PSX and I can see the microSD in Windows.

I tried v1.0.2 again as well and that seems to work now too so I'm not sure what happened with that.
Previously I was using v1.0.0 and having the issues with the LED turning off.

I'm not familiar with pico debugging but it seems like the inverted build is working best for me.

dangiu, 2023-01-23:
Awesome, the LED issue was fixed in between versions. I don't know what happend that caused the inversion, they likely changed something in manufacturing.
Anyway I will now close the issue since the problem has been solved.

closed 2023-01-23 by dangiu