dangiu/PicoMemcard #38: Rp2040 zero with another led

Alex12DEUT, 2023-01-07:
Hey
I received a new batch of rp-2040-zero , i think they have modified the led reference , or maybe just a mistkake from factory
But the red and green colors are "reversed" , when all is fine the picomemcard will show a fixed red color ( instead of green ) , and when it have something wrong it will have green color blinking ( instead of red ) but all is working fine
All the small batch is like that

have someone see that before?
maybe a small uf2 update will fix it? Maybe need inverse the values for red and green 

thx

dangiu, 2023-01-13:
That is one of the most bizzare thing I've ever heard! In the next days I'll compile a modified version as soon as I get a chance so you can try it out :)

Alex12DEUT, 2023-01-13:
Ok thank you i will tell you if it work for picomemcard and picomemcard+ If just inverse code for red/green in your code

dangiu, 2023-01-18:
[InvertedRedGreenBuild.zip](https://github.com/dangiu/PicoMemcard/files/10443507/InvertedRedGreenBuild.zip)
Here are the build with the inverted led both for picomemcard and picomemcard+.
Let me know if it works

Alex12DEUT, 2023-01-20:
Your files are in .elf extension , not .uf2 i cannot put them on my rp2040 :(

dangiu, 2023-01-20:
[PicoMemcard.zip](https://github.com/dangiu/PicoMemcard/files/10464148/PicoMemcard.zip)
My bad, here are the .uf2

Alex12DEUT, 2023-01-20:
i have test for both it ok for the led , now its green when all if fine and red where it have something wrong thx :)
but when make the test i have find another issue with the picomemcard ( not the + ) when you copy a save from normal card to picomemcard ( with ps1 or a ps2 ) if the file is too big its fail ( 5 blocks )
i have test to make a memcard.mcr with 15 block of save ( full ) on computer and inject it , picomemcard see all save , and if try to delete one = fail
i think it have a memory management problem in the code of picomemcard :/

dangiu, 2023-01-23:
Thanks for the heads up, I'll look into it

Alex12DEUT, 2023-01-23:
i have find another thing with the picomemcard ( not + ) when you copy a ''big save '' , like theme park world game  ( 5 blocks )   from ps1 or ps2 browser from genuine sony card to picomemcard , the copy fail

seanchapman, 2023-02-03:
I've added a switch to invert the colours in #41 BTW :)

Davis-A, 2023-07-07:
Looks like i've also got the inverted colours!  I've never soldered before so i thought i must have screwed something up and was looking for information on errors, it started working and i found this issue.  

Here's a video: https://www.youtube.com/watch?v=m8Bqp1F9n0Q

PSXmanCZ, 2024-07-03:
Hello

Please I would like to ask on the PCB board there are three points for LEDs.  And I didn't find any more precise information or photo about it. Can I ask you what type of LED is suitable for this type of project? 

[image]

Thank you
And I must thank the author for the great work I like this project very much.