dangiu/PicoMemcard #40: PMC+ Doesn't work with image above 10.MCR

LobAnovich, 2023-01-26:
When trying to create more than 10 memory card images on PicoMemeryCard+ via SPS1(One), memory cards higher than 10.MCR number are not created.  In the case of creating an image of a memory card 11.MCR and higher through a PC, when you try to switch further than 10 memory cards, the switch does not occur.  The indicator shows that the 10.MCR card is the last one in the list

dangiu, 2023-02-05:
Okk I'll look into it, thanks for the update!

realJoshByrnes, 2023-03-19:
Able to reliably reproduce this issue which limits PicoMemCard+ to 10 memory cards.

Memory cards 0.MCR to 9.MCR work fine.

memcard_manager_create(10.MCR) != MM_OK

realJoshByrnes, 2023-03-19:
Additionally, once the 10.MCR is created, future attempts to create the 10.MCR fail due to it existing and being invalid.

whitezombie2000, 2023-03-23:
I can confirm all the above behavior with a PicoMemeryCard+.

realJoshByrnes, 2023-03-24:
@whitezombie2000 Great feedback, which has identified another bug - https://github.com/dangiu/PicoMemcard/issues/56

I have added firmware which is compatible with PMC+ for Pico/Pico W (not the rp2040-zero) and has the creation fix at https://github.com/realJoshByrnes/PicoMemcard/releases/download/v1.0.22/PicoMemcard.zip if you'd like to try it.

Whilst it should allow you to create more memory cards, it doesn't change the sorting order for cards.

whitezombie2000, 2023-03-24:
Sorry I can't test it because I believe mine is built with the rp2040-zero.

realJoshByrnes, 2023-03-24:
@whitezombie2000 No problems. It's probably worth subscribing to issue #56 also, which is the order bug you've identified.

I've submitted a PR #55 which, if merged, will fix the card creation issue, just waiting on a response from @dangiu

whitezombie2000, 2023-03-24:
I just tried your file so it must be based off of the rp2040. I was able to create 12 memory cards.

LobAnovich, 2023-03-24:
Due to the small size and the ability to place the entire project inside the case of the original ps1 memory card, I had no doubts about choosing the rp2040.  Both my DIY picomemcard + and the card from China have a problem

LobAnovich, 2023-03-24:
> I just tried your file, it worked without any issue, so it must be based off of the rp2040. I was able to create 12 memory cards.

Only 12?

whitezombie2000, 2023-03-24:
> > I just tried your file, it worked without any issue, so it must be based off of the rp2040. I was able to create 12 memory cards.
> 
> Only 12?

Is 20 enough for you? :)

realJoshByrnes, 2023-03-24:
> I just tried your file, it worked without any issue, so it must be based off of the rp2040. I was able to create 12 memory cards.

@whitezombie2000 It's based off PicoMemCard+, but I haven't got an rp2040-zero to test.

> Due to the small size and the ability to place the entire project inside the case of the original ps1 memory card, I had no doubts about choosing the rp2040. Both my DIY picomemcard + and the card from China have a problem

@LobAnovich The rp2040-zero is a nice size. Are you building the image yourself? Feel free to start a new issue and we can see if we can work out what it is @LobAnovich as I have had success with Pico / Pico W / PSXMemCard

> Only 12?

I believe @whitezombie2000 meant they could create 12 successfully, where they couldn't previously.

whitezombie2000, 2023-03-24:
> > I just tried your file, it worked without any issue, so it must be based off of the rp2040. I was able to create 12 memory cards.
> 
> @whitezombie2000 It's based off PicoMemCard+, but I haven't got an rp2040-zero to test. Did the colored LEDs work?

Yes I know the China one is based off the PicoMemcard+ but I wasn't sure if they were using a rp2040 or rp2040-zero in their design. Since your file worked it must be using a rp2040 in their design. Yes the LEDs work.

Hmm...after creating 20 cards it's defaulting to card 7.MCR upon the first inserting into the PS1. This must be related to the order issue.

closed 2023-03-24 by dangiu

realJoshByrnes, 2023-03-24:
@whitezombie2000 Remembering the last card was a precious commit. PR relating to creating new MCs has now been merged (fixed on future builds)

PR #55 (2023-03-24): Card creation fix
At present, it's almost impossible to have more than 10 memory cards without a PC. This PR fixes the issue where 2 digit (eg. "10.MCR") filenames are not created correctly, and also fixes the (unreported) issue where a file could not be created if one of the previous files were corrupted (eg. wrong-size).

Basically, this patch will attempt to brute force the creation of a file, until it finds a filename which does not exist. This may take a short while if you have thousands of cards, and may cause issues if exceeding the filename limit after 9999999999999999999999999999.MCR  [... cut]