dangiu/PicoMemcard #46: PSXMemCard

realJoshByrnes, 2023-03-13:
I was having a little bit of a look around and found [PSXMemCard](https://www.aliexpress.us/item/1005005071752501.html), a commercial pre-built implementation of this software
* [User Manual](https://www.bitfunx.com/wp-content/uploads/2022/12/psxmemcard-usemanual.pdf)
* [TradeMark Application](https://branddb.wipo.int/en/quicksearch/brand/US502022097654216)

I guess I was just curious if this is related to the project, or just using the code from the project?

dangiu, 2023-03-14:
Yes in their page it clearly refers to this project... I didn't know they were trying to trademarke it... A bit scummy if you ask me

realJoshByrnes, 2023-03-14:
I agree, it does seem scummy. But to be fair, the license is permissive, and they're trying to register "PSXMemCard" rather than "PicoMemCard", and it's also just the name rather than the device.

I guess I was just wondering if they contribute back to the project, in a similar way that [BlueRetro](https://github.com/darthcloud/BlueRetro) seems to have set-up. I've ordered one from RetroScaler to have a look.

dangiu, 2023-03-15:
Oh! If they are just registering the name I have nothing against that.
So far I have not been contacted by anybody so I don't think they are interested in contributing.

realJoshByrnes, 2023-03-15:
I was just curious, that's all. If you'd like I'm happy to share info regarding the device once I get it (from China, could take a while!).

To be honest, I've been wanting to make an SD to MemCard project for a while (years), but I was considering doing so in Typescript. With the cost of the rp2040, a C/C++ project is probably the best way to go (was going to use an ESP32 previously), you've done a great job with this project and I'll probably contribute more in the near future.

realJoshByrnes, 2023-03-16:
Since you have one, would you care to share your thoughts on the PSXMemCard @LobAnovich ? Obviously it looks good, but how does it function? Does it have any external buttons for updating etc?

LobAnovich, 2023-03-16:
The Chinese pico memory card is 100% created by the picomemcard + project with the same features and disadvantages.  It does not have a screen, mechanical buttons for switching memory card images and it does not remember the last active memory card image

realJoshByrnes, 2023-03-22:
I've just received the PSXMemCard in the mail, and I'm pleasantly surprised.

I was expecting a MicroUSB, it has USB-C.
I was expecting no way to update, it has a (hidden) button next to the USB port which works in the same way it does on a Pico.

To press the button, you'll need a a SIM card removal tool or similar.

```
RP2040 - Revision B2
25016JVS1Q 2MB Flash Storage
```

realJoshByrnes, 2023-03-22:
Image of the boarcd: https://photos.app.goo.gl/x3GLaLk12rBLKvBS8

The following parts are slightly cropped:
`SW1` Micro-switch that enables firmware upload
`P1` USB-C Port
`D2` Red LED (Light Emitting Diode)
`J9` MicroSD reader

dangiu, 2023-03-24:
That seems to be pretty well made! I believe that some hardware changes will need to be implemented for the project to support PS2 memory cards but hopefully they'll make new revisions.

amak79, 2023-05-22:
The PSXMemCard would need the `PicoMemcard+RP2040-Zero.uf2` image when updating?

realJoshByrnes, 2023-05-22:
> The PSXMemCard would need the `PicoMemcard+RP2040-Zero.uf2` image when updating?

Hi @amak79, I think you're after the [PicoMemcard+.uf2](https://github.com/dangiu/PicoMemcard/releases/download/v1.0.2/PicoMemcard+.uf2) image for this device.

amak79, 2023-05-23:
Thanks @realJoshByrnes. Would it be possible to update the device with the wrong image, and if so would it be recoverable?

realJoshByrnes, 2023-05-23:
@amak79 If you have any issues, you can hold the BOOTSEL button in when you apply power, and reflash.

dangiu, 2023-05-23:
I have no clue, it would depend on how they wired the components together... Do they mention it on the product page? Anyway you can try the RP2040-Zero and if it doesn't work try the Pico one... It shouldn't really be a problem.

Ceep0rt, 2023-10-02:
Good day! I got a curious situation with PSXMemCard. Just purchased one from AliExpress and after trying to boot into memcard menu my PSone just got black screen and after some moment rebooted. Just to specify, after inserting for the first time PSXMemCard and trying to boot PSone into the Memory Card menu, the menu (where you choose between Memory Card and CD Player) itself showed up for a couple of seconds and before I could even press any button the black screen came up, some seconds later PSone rebooted and the only thing I got was the main white screen with SONY logo, without music. From now on when I'm trying (with or without PSXMemCard) to turn on the PSone, the first thing I see is a black screen for like 10-15 seconds, then the SONY white logo appears without music and stays like this as much as you want.

Connecting to PC, the only thing you can explore is a 0.MCR file.

I disassembled my PSone for the first time, previously it belonged to the friend of mine, and discovered some modifications, not sure what kind. I assume something to bypass the original CD requirements.

Any idea? Any suggestion? 

- [image]
- [image]
- [image]
- [image]

dangiu, 2023-10-09:
Sorry but I have no idea... I'm not associated in any way with the products on aliexpress. In addition the modification could have unforseen effects... Just to confirm, you have used it before (without aliexpress picomemcard) and it worked fine?
I'll leave this issue open for a bit to see if anybody has had similar experiences and can answer but my feeling is that this is not related to the project unfortunately.