dangiu/PicoMemcard #47: Data Loss

realJoshByrnes, 2023-03-18:
When trying to sniff, it seems like data is disappearing and out of sync... I thought it was just my code, but it happens on yours, too.

Here we have 0x01 (Controller) 0x42 (Status)
We get a reply of 0xff, 0x73 (Device ID: 0x7 - DS Analog, Length: 0x3 UInt16s), 0x5a (Header End) - Perfect.
For the data section, we get 0xffff66 - expect something like 0xffff6661647b (as received second time)

```
0142000000 0142000000000000 0142000000...
ff735affff66 ff735affff6661647b ff735a...
```

        01      ff
        42      73
        00      5a
        00      ff
        00      ff
        01      66
        42      ff
        00      73
        00      5a
        00      ff
        00      ff
        00      66
        00      61
        01      64
        42      7b
        00      ff
        00      73
        00      5a

The same thing is happening in my own code (using psxSPI.pio):
````
RX: 01 42 00 00 00 01 42 00 00
TX: FF 73 5A FF FF 66 FF 73 5A
RX: 00 00 00
TX: FF FF 66
RX: 01 42 00
TX: 61 61 FF
RX: 00 00 00
TX: 73 5A FF
````

Initially I thought it might be because I was using printf() in the interrupt, but it seems to be happening with your example, too. Are you able to sniff without issues?

realJoshByrnes, 2023-03-19:
Just had a thought that maybe it's only Pico W
Further testing that the same happens on Pico (non-W)

dangiu, 2023-03-19:
Did you use the sniffer example to sniff? I don't quite remember how it set it up (I added it to the repository mostly as historical material) but I remember that I was having trouble when printing during interrupts or "during capture". To get reliable output I needed to store the data received inside an array and then print it all after the execution. A sort of post-mortem analysis

realJoshByrnes, 2023-03-19:
Yeah I tried that example, as well as my own. Submitted PR in #51 for the proof-of-concept projects.

It would be great to have the discussions tab enabled to discuss possibilities.

dangiu, 2023-03-19:
Ok I'll look into setting it up, thanks for the quick feedback!

realJoshByrnes, 2023-03-23:
> Ok I'll look into setting it up, thanks for the quick feedback!

The issue appears to be in cmd_reader PIO program. I don't think you need the [bit counter](https://github.com/dangiu/PicoMemcard/blob/pmc%2B/release/psxSPI.pio#L30) when autopush is enabled.

Using the following psxSPI.pio is in sync and works for sniffing (doesn't ACK on purpose)
https://github.com/realJoshByrnes/PicoMemcard/blob/641992ca1d0fb8ca68476bb61867e36dda61b4e5/pio/psxSPI.pio#L28

realJoshByrnes, 2023-03-28:
@dangiu It sounds good as long as we can choose to send the ACK (so we don't send after the last byte)