dangiu/PicoMemcard #56: Memory Card switching order

realJoshByrnes, 2023-03-24:
As per https://github.com/dangiu/PicoMemcard/issues/40#issuecomment-1482001010

> it will load card 0.MCR as it should, first switch it will load 1.MCR, on the 2nd switch it will load 10.MCR, on the 3rd switch it will load 11.MCR, on the 4th switch it will load 12.MCR, on the 5th switch it will load 13.MCR, on the 6th switch it will load 14.MCR, on the 7th switch it will load 15.MCR, on the 8th switch it will load 2.MCR, on the 9th switch it will load 3.MCR, on the 10th switch it will load 4.MCR, on the 11th switch it will load 5.MCR. on the 12th switch it will load 6.MCR, on the 13th switch it will load 7.MCR, on the 14th switch it will load 8.MCR, on the 15th switch it will load 9.MCR.

PMC+ is sorting images alphabetically as per https://github.com/dangiu/PicoMemcard/blob/pmc%2B/release/src/memcard_manager.c#L93 which is problematic for numbering (1, 10, 11, .. 19, 2, 20, 21...)

dangiu, 2023-03-24:
🤦‍♂️🤦‍♂️🤦‍♂️ nice catch!

wired-filipino-owl, 2023-06-10:
Has this been resolved with the new release 1.03, or are we waiting for testing?

dangiu, 2023-06-12:
Still needs to be fixed, I forgot about this

whitezombie2000, 2023-06-12:
I attempted to test this issue this last night before dangiu posted that is was not fixed yet. I have two clone Pico cards from China. I flashed v1.03 and the card would no longer function when inserted into my PS1. It will not read nor will the LED light come on. It works when inserted into the PC, I can read the files, and put it into flash mode. I flashed v1.02 and it worked just fine in the PS1. I also flashed realJoshByrnes forked v1.0.22 and it worked fine. I noticed the file size increased a lot with v1.03, its 646 KB, v1.02 is 166 KB, and the forked version is 180 KB. I don't know if something got messed up with v1.03 but thought I would mention the issue I was having.

wired-filipino-owl, 2023-06-14:
@whitezombie2000 this looks worthy of opening a new issue

When I get time, I'll also try with my clone card.

whitezombie2000, 2023-06-14:
@wired-filipino-owl Sounds good. I have two of these black ones. 

[image]

dangiu, 2023-06-14:
@whitezombie2000 you are right. Wierd, i tested it on my pico and it worked... Right now I'm on vacation but once I get back I will try to figure out what is going on... With this release there have been quite a few PR merged and I may have missed something that resulted in the huge increase in code size. My guess is that it may be something related to PR #53 since it applies only to Pico and not to RP2040-Zero

whitezombie2000, 2023-09-06:
@wired-filipino-owl Did you ever test your cards to see if you could replicate the issue?

@dangiu Did you ever get a chance to look at the code and figure out why the code size increased so much? 

Thanks everyone!

realJoshByrnes, 2023-09-06:
> @dangiu Did you ever get a chance to look at the code and figure out why the code size increased so much?
> Thanks everyone!

I believe the code increase in size may have been #53 (this was expected) as some of the wireless stuff has to be included for the Pico_W support (even though we're not using wireless, the LED etc is managed by the wireless chip).

whitezombie2000, 2023-09-06:
@realJoshByrnes Did you have a clone to test?

wired-filipino-owl, 2023-09-06:
> @wired-filipino-owl Did you ever test your card to see if you could replicate the issue?
> 
> @dangiu Did you ever get a chance to look at the code and figure out why the code size increased so much?
> 
> Thanks everyone!

@whitezombie2000 I'm away on summer vacation right now. Thanks for keeping the issue active! If this is still ongoing when I return, I will test the latest build with my clone Pico memory card.

dangiu, 2023-09-07:
@whitezombie2000 no I did't look into it yet but I agree with @realJoshByrnes . I'm also way on vacation and will be back working on the project at the end of september :)

wired-filipino-owl, 2023-10-01:
Tried 1.0.3 out on the Aliexpress card. Used Pico firmware, since that worked previously. I'm guessing the Zero firmware has a different pinout.

Results:
Card is unresponsive with SD card inserted. Status LED does not light up.
With SD card removed, status LED will come on after inserting card. LED is pink, not red. Opening up the case reveals that the status LED is an RGB LED, complete with three resistors running to it??? I tried the Zero firmware (1.0.2) and the card was unresponsive.

Went back to 1.0.2 Pico firmware and the card worked again.

I think I'll stay on 1.0.2 for now.

dangiu, 2023-10-09:
Thanks for the testing, maybe I'll write a comment about this on the main page. I'm not involved in any way with those chinese products so I don't really know how they were engineered and which pinout they use

whitezombie2000, 2024-07-02:
Just a heads up that there is a new fork created and it works pretty good with the Bitfunx Memory card. The LEDs are all working, changing color, and Game ID has been implemented. I have also spoken with MrPalloncini on another platform and he's going look into the alphabetical issue. 

https://github.com/MrPalloncini/PicoMemcard

whitezombie2000, 2024-07-10:
I highly recommend you check out this fork. Game ID works great with the X-Station, PSIO, FreePSXBoot, or a cart flashed with Unirom. He also got rid of the alphabetical\decimal ordering issue altogether, got rid of the LastMemcardIndex.dat, and moved everything into the index.txt file. It works really well.

https://github.com/MrPalloncini/PicoMemcard/issues/1#issuecomment-2216972030

whitezombie2000, 2024-07-13:
@dangiu @realJoshByrnes Can either one you help MrPalloncini out and get Unirom working? He did a fantastic job making it work with PSIO and the X-Station.

https://github.com/MrPalloncini/PicoMemcard/issues/1

whitezombie2000, 2024-07-30:
I highly recommend you check out this fork. Game ID works great with the X-Station, PSIO, FreePSXBoot, or a cart flashed with Unirom. He also got rid of the alphabetical\decimal ordering issue altogether, got rid the LastMemcardIndex.dat, and moved everything into the index.txt file. I did a lot of beta testing for MrPalloncini, discovered lots of bugs, he got it stable, and it works really well.

https://github.com/MrPalloncini/PicoMemcard

JayroJones, 2024-08-04:
> I highly recommend you check out this fork. Game ID works great with the X-Station, PSIO, FreePSXBoot, or a cart flashed with Unirom. He also got rid of the alphabetical\decimal ordering issue [...]

Dude, I was starting to think my PSxMemCard was faulty, because the v1.0.3 firmware from this repo gave me NO LED at all and the card just acted dead. This new fork totally fixed all my issues! (v1.0.2 from here would work, but hot-plugging it was hit or miss weather or not it would re-load the contents, and the LED was stuck on RED color.)

Thanks again for this! LED is green now, and loads my save files every time with no issues! :D

whitezombie2000, 2024-12-28:
Hey Everyone,

There is a new a firmware for Pico cards based on SD2PSX and it works with PS1 and PS2! You can emulate a PS1 memory card and a PS2 memory card with GameID. The MMCE (Multipurpose Memory Card Emulator) group set out to create a new protocol. I did some brief testing with it on the PS1, creating some PS1 GameID memory cards, did some testing on PS2 creating some 8MB PS2 GameID memory cards, and even set up a Free McBoot card. All this was done using a PSXMemCard Gen1. MMCE seems to be the future, has active development, and I encourage you try it!

Link to the 1.0.0-beta10 firmware for PicoMemcard+, PicoMemcardZero, PSXMemCard Gen1, and PSXMemCard Gen2.

https://github.com/sd2psXtd/firmware/releases

Here's an invite to the PS2 Scene Discord and you can discuss in the sd2psx channel.

https://discord.com/invite/B3dCMsdbBe

Here's a quote from the dev bbsan2k in the sd2psx channel.

"Hello everyone. In the last couple of months, the MMCE (Multipurpose Memory Card Emulator) group worked on giving you the ultimate MMCE experience with your already existing hardware. The MMCE group includes (but is not limited to) your friends from 8BitMods and sd2psXtd developers.

This public OPL beta will allow you to use the SD card in your MMCE device (like MemCardPro 2, sd2psx and PicoMemCard+) as a data source in OPL, effectively enabling game playback.

It is compatible with the following firmwares:
MemCardPro 2: 1.3.0-beta2
sd2psx / PSXMemCard Gen2: sd2psXtd 1.0.0-beta10
PicoMemCard+ / PSXMemCard: sd2psXtd 1.0.0-beta10"

Here is a link to the 8Bitmods Discord if you are looking for the MemCardPro 2 firmware.

https://discord.com/invite/F3aKahj

leafy-psx, 2025-08-10:
I think the numbering is normal for ASCII, you will see the same thing when listing a directory on Linux. Possibly using 00, 01, 02 would  be a quick hack.