dangiu/PicoMemcard #57: Controller does't work with picomemcard

vini-XD, 2023-04-07:
I was using picomemcard for a couple of days normally with psone (slim), suddenly after i deleted a game in the menu and restarted the game the controller stoped working. So everytime the picomemcard is on (slot 1 or 2), the controller stops working, the led still turns on but does't work, that purple pointer to select disappears in the memory card menu everytime the picomemcard is pluged.
I tried changing slots and another controller, nothing works...
Any idea?

vini-XD, 2023-04-17:
Ok. When I have the time, I'll try to disassemble the picomemcard and solder all again, I'll post here if it worked

realJoshByrnes, 2023-04-30:
The other option would be trying a fresh install; Backup your save games to another device and restart PMC.