dangiu/PicoMemcard #60: Psx controller emulation.

DiktatorShadaloo, 2023-06-14:
Hi, I was looking if there was any project to emulate a ps1 or ps2 controller using a raspberry pi pico to make an arcade stick and I found this repository whose description says that it serves to emulate a psx controller but the readme is focused on emulating the memory card and there is no info on how to emulate the control or diagrams of the buttons. Is it possible to do it or it isn't implemented yet?

Thanks in advance.

wired-filipino-owl, 2023-06-14:
This project is focused on reading the controller inputs to switch emulated memory cards. That may be the reason it came up in your search. I don't believe there are any plans to do controller emulation, as that is significantly outside the focus of the project.

If you wanted to make a PS1 controller emulator yourself, the PS1/PS2 controller protocol is very well documented online.

https://gamesx.com/controldata/psxcont/psxcont.htm
https://psx-spx.consoledev.net/controllersandmemorycards/

I also found a half-finished Arduino (ATmega328p) project from a decade ago
https://web.archive.org/web/20130523144355/http://novemberbravo.50webs.com/projects/avr_dsc/avr_dsc.html/

dangiu, 2023-06-14:
Hi, it is very well doable, initially I began the project by emulating a PSX controller. If you know how to write some basic code it will be quite easy. In the `poc_example` folder there is an example of PSX controller emulation, try to take a look at that.
Right now I'm on vacation but once I get back we can setup a discord call if you are not able to do it by then :)

dangiu, 2023-06-19:
The code is a bit outdated so it probably will not compile straight away and might need some fixes but the basics are there :)