dangiu/PicoMemcard #61: [Suggestion] Is it possible/ too hard to integrate with xstation or other ODE?

almircanella, 2023-06-18:
I mean, There is the memcardpro that auto load specific memcard for specific game running on Xstation, and I ask if that is black magic or if it is in your plans to implement that feature in the future.

It will be very nice.

Thanks in advance.

dangiu, 2023-06-27:
It should be possible. Right now, for example, the software is capable of reading which game is being launched with uniboot... I belive Xstation does something very similar to that but I'm not sure because I don't own one. Maybe somebody with more expertise can chime in

ramapcsx2, 2023-06-28:
It is the same mechanism. If you support Unirom, xStation also works.. Should work anyway :)

dangiu, 2023-06-29:
Awesome, I'll close this issue when support is fully added :)

almircanella, 2023-07-03:
Another tip: mind the psycho mantis thing...If the feature could be activated pergame, it would be better.

 I dont know other games that have save game conversation among different titles, but I supose there is some.

Dont know also about patched, translated, romhacked games if mess things up or whatever.

Anyways, thanks for looking into it.

atds55, 2023-07-03:
Memcard pro and SD2PSX do this and I think people don't have a problem with this function.

airainchoc, 2024-02-28:
Has someone looked at this (maybe a fork) ? Would be awesome

MrPalloncini, 2024-06-29:
I've been able with my friends to get it working. I will fork this project and upload the code once i'll make few more tests. In the end every game will create a new .MCR file based on what SYSTEM.CNF says and stores it inside a file called index.txt. Keep in mind that this will create a different memory card even if the game is the same but from a different region and if the same game has multiple discs since the ID for each one is different.

@almircanella
I don't think this would be possible for few reasons. First you need an entire database of all Konami's games supported by MGS stored in the SD card, then the code would be slower since for each game it need to assign the same memorycard. The check if the game exists in the index file is already slow enough. In the end this doesn't make much sense. What i can suggest you is to boot the game, pick the MCR file assigned and transfer the saves using third party tools.

almircanella, 2025-01-16:
> I've been able with my friends to get it working. I will fork this project and upload the code once i'll make few more tests. In the end every game will create a new .MCR file based on what SYSTEM.CNF says and stores it inside a file called index.txt. Keep in mind that this will create a different memory card even if the game is the same but from a different region and if the same game has multiple discs since the ID for each one is different. The fork will also fix the latest code, making compatible with the Aliexpress Bitfunx memorycard. The setting is stored as an #ifdef in the config.h file
> 
> [@almircanella](https://github.com/almircanella) I don't think this would be possible for few reasons. First you need an entire database of all Konami's games supported by MGS stored in the SD card, then the code would be slower since for each game it need to assign the same memorycard. The check if the game exists in the index file is already slow enough. In the end this doesn't make much sense. What i can suggest you is to boot the game, pick the MCR file assigned and transfer the saves using third party tools.
> 
> EDIT: here is my fork: https://github.com/MrPalloncini/PicoMemcard/releases/tag/Latest as said in the release, might not work with your hardware since i don't have one for testing, i will make a pull request after some more testing, but right now seems to work just fine. I would like to hear your experience with it.

So, sorry for the delay, you know, family. well, i used your fork, the latest updates, so nicely worked: 

-also, i've refer some (3 or 4) gameID (konami ones) to the same vmc img on the index list, so when a i launch some of the games, the same memorycard is loaded just fine, the way i wanted to.

-there's also sd2psx firmware that now also works with picomemcard, but i havent (and problably wont) try it.