dangiu/PicoMemcard #62: Tiny2040 Support

matteog9, 2023-07-11:
Hello.

I have a question. I flashed the PicoMemcard firmware to a Tiny2040, by Pimoroni. However, when I insert it into the USB drive, I hear the usb inserted chime, but no removable storage drive appears. I don't know if the Tiny2040 differs from the RPI 2040 Zero. 

Thanks!

NB: Tiny2040 Product Page: https://shop.pimoroni.com/products/tiny-2040?variant=39560012234835

dangiu, 2023-07-18:
I'll look into it, but first of all I need more info. What version did you install on it (the "plus" version or the regular one)?

matteog9, 2023-07-18:
Thanks! Sorry, I should've specified which version in the original post. I installed the Plus version onto the Tiny2040.

Thanks again.

dangiu, 2023-07-25:
No they have a different pinout, unfortunately I need to make a build specific to the Tiny2040, I'll try to make a build of the old stable release farily quickly.
Some functionality of the current RP2040-zero version should still work, in particular you should be able to see the sd card popping up in your computer, if it does not there is probably some hardware problem or maybe the SD card is not formatted correcly, try to check that one out in the meantime.
As a reminder, the "plus" version requires the additional SD card reader

matteog9, 2023-07-25:
Thanks! I will try that. I only got as far as flashing the uf2 file to the Tiny2040. I hope there aren't a lot of changes to make to the code.

I don't know if this matters, but Pimoroni make the Tiny2040 in both 2 MB or 8 MB of QSPI flash supporting XiP. 

Thanks again.