dangiu/PicoMemcard #67: Please insert disk into usb drive error after installing picomemcard software on picomemcard +

D3iv, 2023-08-25:
I've tried solving this problem myself to no avail, couldn't find anything on the internet regarding this issue. Checked all of my solderjoints, perhaps I'm missing a library or a driver of some sorts?

ste94pz, 2024-05-09:
Same problem here on rp2040-zero, microsd formatted correctly, recompiled with develop pico-sdk. I'm using this microsd module with the AMS1117 regulator bypassed and connected to the 3.3v pin of the rp2040-zero.
[image]

dangiu, 2024-05-10:
To me it seems like your setup is correct.
You have included the fix mentioned in [raspberrypi/pico-sdk#1304](https://github.com/raspberrypi/pico-sdk/issues/1304) correct?
Maybe you could try to swap the microsd module for a different one just to make sure that it's not faulty.