dangiu/PicoMemcard #68: [INFO] Quirks of the Aliexpress PicoMemcard

wired-filipino-owl, 2023-10-01:
@dangiu, please pin this issue if you think it's worthwhile to learn more about this board.

Owners of this device, please reply with any additional information you have about this board!

BitfunX on Aliexpress sells a PicoMemcard ready to go, including a no-name 512MB microSD card. Here is some things I was able to figure out about this device.

The status LED is RGB. I can see three lines going to it on the board, plus ground I assume. LED is labeled D2. See the photos of the board.

[image]

[image]

Now let's see what GPIO those pins correspond to, I counted 5 pins from the left edge, which would make the lines connected to physical pins 35-37. Those correspond to GPIO 23-25. See the pinout from the RP2040 datasheet:

[image]

Quality of the device:
I think it's worth the $15 USD. Winbond flash chip, quality molded plastic case, and push-eject SD card slot. Also, the BOOT/BOOTSEL button is accessible without taking apart the case, just use a paperclip or similar sized object to press the button.

wired-filipino-owl, 2023-10-01:
I was able to flash the MicroPython firmware and run an RGB example program on my PicoMemcard. The pins hooked to the RGB LED are indeed GPIO 23-25.

dangiu, 2023-10-09:
Great investigation!

dangiu, 2023-10-09:
As @wired-filipino-owl pointed out in #56 Release 1.0.2 for the Pico seems to work correcly on this board

MrPalloncini, 2024-06-29:
The release 1.0.3 work with the Bitfunx memorycard without any issue, the problem was that the new led code for the pico_w broke the compatibility. i've been able to restore it by adding some #ifdef to the config.h. It's also worth noting that on the back of the card there are 3 pins exposed which are the pio0, pio1 and gnd. They are useful for debugging purpose since i used it all the time to print .
[image]

amak79, 2025-02-11:
Does anyone know of a source for the boot button (`SW1`)? Mine snapped off with little force and now it can't be updated unless I temporarily bridge the two pads.

wired-filipino-owl, 2025-02-14:
@amak79 as long as the pads are still intact, you should be able to find an equivalent search at an electronics distributor such as Mouser. Here's a link with some filters to get you started: https://www.mouser.com/c/electromechanical/switches/tactile-switches/?mounting%20direction=Right%20Angle&mounting%20style=SMD%2FSMT

I'll add another comment if I find the exact switch.

amak79, 2025-02-14:
Thank you very much.

I initially looked at [JLCPCB parts](https://jlcpcb.com/parts) but was overwhelmed with the thousands of buttons and switches.

amak79, 2025-02-18:
I couldn't find the switch on Mouser. I asked Bitfunx and surprisingly they replied with the link to the correct switch: https://www.lcsc.com/product-detail/tactile-switches_sm-switch-smg-10-h035b0_C29781355.html