dangiu/PicoMemcard #70: RP2040 generic

madmai, 2023-10-27:
I use this how rp2040 [https://imgur.com/8IVB1Z4](url) and no Works, sometimes blue led blinking not ever
and this works randomly sometimes green led fixed sometimes blinking this is original rpi pico[https://imgur.com/HICGgex](url)

what can i do for 2 versions works, pi pico original sometimes works, but the black one, never works, anyhelp is apreciated, thx

dangiu, 2023-10-30:
To solve this I will need more information:
- Which version of picomemcard are you trying to use?
- Which devices are you using with it? You mentioned one original pico and one black, I need more informations about the second one.
- Have you checked the soldering joints with a continuity tester to make sure they are good?

madmai, 2023-10-30:
[https://imgur.com/NCuncBR](url) now i'm using this, and yesterday works, but now blinking led green in original pi pico, with release pico 1.0.2, before i test too 1.0.3 pico, but now version i use is 1.0.2 pico, i have those too [https://imgur.com/ovCDs9p](url) i build 2 memcards adapter to plug in psx, and when i have i test 3 models, with 1.0.2, black one, purple on and pico original, i tell something when i have, i thing in the one i have build the conectivity is not the problem sometimes works, sometimes blinking in pico original, now is blinking, if you want tellme in pm too, thx, for your attention

dangiu, 2023-10-30:
I'm sorry but I'm having trouble understanding everything. From what I can tell if you are experiencing intermittent issues (working for a bit than not working) I think the most likely cause is some sort of bad connection

madmai, 2024-04-25:
I think i need 2 red cables vcc, it's correct? someone can explain me, how to feed vcc 3.3v micro sdcard , and how to feed pi pico vcc voltage, please explain me scheme of feed voltage map in pico drive and micro sd, i think this is the problem, only voltage feed schema, for disolve my dudes... thanks for advance to all people, thx again

dangiu, 2024-05-03:
I believe the adapter from [this picture](https://imgur.com/NCuncBR) needs 5v on VCC (and does its own conversion down to 3.3V) so you need to connect the Pico VSYS (pin 39) to it. Question: the Pico is powered by the PSX right?

madmai, 2024-05-03:
i think the psx 3.3 v memcard port not works properly only do 1.5v(maybe with tests brokes), i think i repaired to 3.3v memcard port now, but tell me how i have to connect pi pico voltage and sd voltage i have others sd ports works in 3.3v, if you can tell me how i have to connect pi pico and sd card, the scheme vcc connections in pi pico, thanks in advance, in the project image, [this image](https://github.com/dangiu/PicoMemcard/blob/pmc%2B/release/docs/wiring-pico.svg), tell me how to connect pi pico to receive 3.3v and how connect out to sdcard port, i don't understand in previous image please, can you specify me, where connect from mc 3.3v to pi pico and sd port voltages vcc port?, first of all, thanks a lot for your attention and replies, thanks again, and my apolyge for my english, i'm spanish.
i wait a reply, thx

madmai, 2024-05-06:
> I have trouble understanding everything but here is what you need to connect: To power the pico: MemoryCard 3.3v <--> Pico VBUS (40) To power the SD module (if it has it's own regulator): Pico VSYS (39) <--> SD VCC To power the SD module (if it needs 3.3v): Pico 3V3(OUT) (36) <--> SD VCC Ground must be in common for everything (GND on PSX, GND on Pico, GND on SD)

ok a lot of thanks, when i have time, i try, again thanks