dangiu/PicoMemcard #71: RP2040-Zero not recognized as a storage device (Windows 11)

Manghao, 2023-11-06:
Hello, I have an RP2040-Zero, I transferred the file PicoMemcard+RP2040-Zero.uf2 on it by pressing well on the BOOT button before connecting it to the PC, I have the notification sound windows which dismounts and remounts the USB device, but I do not have access to the storage of the RP2040. I do have PicoTM Mass Storage USB Device

[image]
[image]

Manghao, 2023-11-09:
I have the same problem on MacOS, the RP2040 disconnects with an ejection error, although it is present but not displayed in storage.

[image]

dangiu, 2023-11-11:
3 things you should check:
- is the SPI sd module installed correclty (wires and connections)?
- is the microsd card inserted?
- is the microsd card formatted in FAT32 (only format currently supported)?

dangiu, 2023-11-21:
Thanks for the info @carmax91. The related PR is not merged yet but I'll patch it myself for the next release at least until it gets merged.

NDR008, 2023-12-09:
Is there a manual way I can work around this issue?

ste94pz, 2024-05-08:
Hi, any news about this problem?

danvx6, 2024-11-15:
I've also run into this issue.