dangiu/PicoMemcard #83: Controller doesn't work when plugged in

Seva167, 2024-07-22:
I have a SCPH-7000 Japanese console and aliexpress Psxmemcard, when i try to plug it in bios acts like controller is unplugged, when already in memory card menu, it still doesn't work.

Is this aliexpress memcard issue or japanese bios issue?
Tried different memcard and controller slots.

Blank firmware still triggers this issue.

realJoshByrnes, 2024-07-23:
I can't imagine that it would be an incompatible BIOS, but it's always a possibility (the same is true for the controller)

What do you mean by blank firmware? Did you flash the device?

Have you tried with/without the SD card? Does this make a difference?

I suspect that the device may be defective, but it could also be a software issue. As we don't know what's on the device, can you try flashing a different version of PicoMemCard on it?

Seva167, 2024-07-23:
Tried most of the versions.
Blank firmware is firmware that disables all gpio and goes into infinite loop.

SD card makes no difference except blinking LED indicating error.
RP2040 appears to be working, and i'm not sure how it is even interferes with the controller.
Even when power pins are blocked it still interferes, maybe some pins are shorted somewhere.

realJoshByrnes, 2024-07-23:
> maybe some pins are shorted somewhere.

Once I read that you're actually installing blank firmware (ensuring you didn't mean a clean PMC install) my first thought was that you have a short.

If you've got a multimeter you could test for continuity between pins.

I'd get in touch with the seller. Hopefully it's just one bad device, and not a bad batch.

Seva167, 2024-07-24:
Looks like in my case pin 2 is shorted with pin 6

Seva167, 2024-08-11:
Got a refund from aliexpress, ordered a replacement, works now.

closed 2024-08-11 by Seva167