dangiu/PicoMemcard #84: What about the LED indication in version 1.0.3?

PSXmanCZ, 2024-09-11:
I looked on various threads to see how it is with the LED indication I found some descriptions that it changes red for orange if I understand correctly to make the unsynchronization more obvious. 

Then I noticed that for some of the newer versions of the RP2024 you applied a swap of red and green. And I must have bought an older version of the RP2040 on Aliexpress this year or they got something wrong in that copy. 

I've always used opaque boxes so I didn't notice it but now I've played around with the design. So that I know the status of the memory card. And now my card is glowing red all the time and working as it should and when something is being saved it glows green. Even the error message indicates green. It's confusing.

I have no programming experience and don't want to burden you. I would just like to ask you which parts of the code I can change for version 1.0.3 and swap the colors (green with red) for my RP2040 model? I guess in the led.c file ? I just don't know which line(s).

[image]

How to compile it into the final .uf2 file I can find somewhere on YT or google. That's how I'm used to educate myself.

Thank you for your great work.

wired-filipino-owl, 2024-09-29:
You're correct that `led.c` is what controls the LED colors. For compiling the project into a `.uf2`, consult the official Raspberry Pi documentation: https://www.raspberrypi.com/documentation/microcontrollers/c_sdk.html#sdk-setup and https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf

There's an updated fork of this project by MrPalloncini that adds some features and bugfixes here: https://github.com/MrPalloncini/PicoMemcard. This includes my PR to add support for the LED on the BitfunX/Aliexpress PicoMemcard device. I believe the LED colors have been changed back to match the README table in that version. As you said you have limited programming experience, I suggest you check out MrPalloncini's version first, before trying to make your own.

Warning: Being Pedantic Below
> The RP2040 is the chip itself, which doesn't change regardless of what RP2040 board you get. What I believe you were asking is what board configuration you have, which is kind of hard to figure out without looking at the board or the store listing.

PSXmanCZ, 2024-10-01:
Thank you for the information. I'm going to study. I know about the MrPalloncini project. But unfortunately he focuses more on Bitfunx and Game ID also got xstation it would be interesting. But unfortunately no relase works for me on my picomemcard+ (PR2040) or on the rasbery pi pico version. With version 1.0.4 the memory card lights up but can't find the MCR image even if I edit the index.txt as instructed. I am therefore referred to the original project. The author of the readme file states that he only tests on Bitfunx and does not have PicoMemCard+. But thanks a lot anyway.

PSX-MAN, 2025-08-09:
It would probably help, but unfortunately I don't have enough experience in compiling code. That's where my efforts failed. It's complex, and I had to install a lot of programs (plugins), even though I went through all the various tutorials. It seemed like it wasn't that difficult, but when you don't know what you're doing and why, it's hard to find where you went wrong when the compilation ends with a generic error.

What would you recommend for compiling? I tried a combination of Visual Studio Code + various plugins. Maybe I'll try again sometime; I don't like to give up. At the same time, I don't want to bother people who are willing to do such amazing projects in their free time. All I need is a little tip, and I'll try to find out more about it myself.

Thank you.