dangiu/PicoMemcard #94: Picomemcard Zero (no + version): problems when copying MEMCARD.MCR file in Linux

madmalkav, 2025-03-08:
If I copy the MEMCARD.MCR file in Linux, it seems to copy OK (same md5 and all) but after you dismount the drive, the file is corrupted. It works OK if I copy the file from Windows.

madmalkav, 2025-03-29:
>   Have you tried manually synchronizing I/O before removing the drive?
> In general files are not moved immediately when we do a copy-paste, and they are actually written at a later stage. Windows and Linux have different policies about when this should happen,  and the "safe eject" feature is there exactly to ask the operating system to hurry up and commit all the postponed operations. Sometimes though Linux takes a long time and you end up disconnecting your device under the impression your files were actually copied, while in fact they were still uncommitted.
>
> Have you tried manually synchronizing I/O before removing the drive?
> In general files are not moved immediately when we do a copy-paste, and they are actually written at a later stage. Windows and Linux have different policies about when this should happen,  and the "safe eject" feature is there exactly to ask the operating system to hurry up and commit all the postponed operations. Sometimes though Linux takes a long time and you end up disconnecting your device under the impression your files were actually copied, while in fact they were still uncommitted.
> You can force a manual synchronization by opening a terminal, typing sync and waiting for it to finish (it will not print anything, but the terminal will not accept new commands until it finishes). After a sync you can safely eject your drive.

I tried and didn’t solve the problem

leafy-psx, 2025-08-09:
Have you copying tried as root or using sudo? Possibly a USB device permission. Possibly need to edit your udev rules to allow it.

leafy-psx, 2025-08-10:
To anyone else with this issue. I had trouble until I;  logged in as root, unmounted the Pico MC, as root mounted it again to /mnt/pmc, as root copied the memcard.mcr. I'm still trying to figure out the best/etc/udev rules.